Ultimaker/Cura #4628: 3.4.1/3.5.1 - issue with "Skin Overlap" feature

amagr0, 2018-10-22:
**Application Version** - CURA 3.4.1 / 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Open Cura Project; analyse the toolpath.

**Actual Results** - Some lines are trespassing from the wall, this seems to be related with "Skin Overlap". Is also strange that is happening in the middle of the part, and not only on the skins parts.  

[image]

If "Skin Overlap" = 0, the results of the cgode generated on 3.4.1 and 3.5.1 are diferrent.
The result on 3.5.1 is strange since the wall have 1.21mm and that is enough for 3 perimeters, but for any reason is generated a "small" wall, assuming that is a gap between walls.

Cura 3.4.1
[image]

Cura 3.5.1
[image]

 

[BUG_1.zip](https://github.com/Ultimaker/Cura/files/2501524/BUG_1.zip)

rburema, 2018-10-26:
Hi @amagr0, From the description of the Skin Overlap setting: 

> The amount of overlap between the skin and the walls. A slight overlap allows the walls to connect firmly to the skin.

To me, it seems like it does what it's supposed to do.

Now you could of course argue that

 - there should be a warning when the overlap is bigger than the wall-size
 - and/or there shouldn't be a difference between layers here.  

... but neither of these, given the definition, seem like large improvements, so I'll still close this issue for now. But if you think those two are important, please reply so we can discuss potentially reopening them, or creating a new issue.

cheers,
Remco.

amagr0, 2018-10-26:
Hello @rburema 
The "Skin Overlap" value is 0.35 mm, and the "Wall Line Width" is 0,4 mm, so how can the lines from the skin overlap trespass the wall perimeter?

Tks

rburema, 2018-10-26:
Hi @amagr0, That is a good point. I'll reopen this issue.

Ghostkeeper, 2019-01-18:
Still reproduces in 4.0.

LipuFei, 2019-01-22:
For dev, JIRA ticket: CURA-6140

rburema, 2019-02-15:
Hi @amagr0,
I took another good look and I found out what the issue is:

The nozzle, while printing the arching lines, stops at 0.35 mm relative to the start of the wall. (Correctly, I think.) You can see that if you have 'travels' enabled in layer-view. However, since the line thickness is indeed 0.4 mm, the material can still 'stick out' of the wall by -(0.4 / 2 - 0.35) = 0.15 mm. In order to guarantee not to have it stick out, you should select a skin overlap of (less than) half the wall-tichkness.

In the process we found that the description of Skin Overlap (and especially '.. Percentage') is confusing at best and probably misleading, so we'll change that at least.
cheers,
Remco.

closed 2019-02-15 by rburema