Ultimaker/Cura #4629: Cura 3.5.1 STL vs Gcode

Thisismydigitalself, 2018-10-22:
**Application Version**
3.5.1

**Platform**
Win10/64/NV750TI latest driver

**Steps to Reproduce**
Slice and then load and parce the same generated gcode back in Cura 3.5.1

**Actual Results**
Following messy print i checked the gcode and it seems there are line/walls issues which are not visible when slicing stl or reloading project but clearly seen on printed part.

**Expected results**
Given having a well tuned printer, Slice > gcode > actual print should look the same. 

**Additional Information**
screenshots attached as well as *.3mf file.
- In slice mode lines looks well defined and smooth
- In Parced gcode screen it is clear lines looks bad - much like the actual print which looks the same.

Sliced STL:
[image]

Parced gcode (Printed part looks the same):
[image]

[351Project.zip](https://github.com/Ultimaker/Cura/files/2502060/351Project.zip)

smartavionics, 2018-10-22:
I think that most of those glitches are due to the buggy wall overlap compensation. When I turn that off the result looks better in a gcode viewer.

Also, don't use the sharpest corner z-seam alignment as that rarely does the right thing unless the object has nice sharp edges. You're much better off using the user specified z-seam hint which will vertically align the z-seam (e.g. set the z-seam X to 100 and Y to 0 and the z-seam will run down one side of your print). 

Hope this helps.

Thisismydigitalself, 2018-10-23:
Thank you. Indeed, it does looks like compensation wall overlap is the culprit. I was under the impression that Cura's Slice/Layer space will reflect the outcome and that the generated gcode will not be any different but this clearly not the case. Cura Slice/Layer Space must simulate the outcome or allow the user to parse the gcode and analyze it when still in Slice/Layer space before saving/sending gcode to printer. no room for surprises.

Thisismydigitalself, 2018-10-23:
Parsing takes time but not that long if you take into consideration Cura's layerview unfaithful representation of the generated gcode and you're just about to start a 5 hours ot longer print job. If one of the settings known to be that buggy, it must be moved to experimental section.

diegopradogesto, 2018-11-08:
If you see this small blobs in the gcode is because the gcode parser has to calculate the width of the line and in some cases the calculation is not really correct (and it takes time). I'm surprised that the blobs in the gcode are exactly the same in the printed part. Could be that they are at the end of the layer?

closed 2020-06-20 by no-response[bot]