Ultimaker/Cura #4630: 3.5.1 - Dual Extrusion - Please revert to the old Prime Tower

amagr0, 2018-10-22:
**Application Version** - CURA 3.4.1 / 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Open Cura Project; analyse the toolpath.

**Actual Results** - The new "Prime Tower" presented on Cura 3.5 is increassing a lot the print time, and also increassing the probability of contamination with the inactive extruder.
Now, for each layer both extruders must complete "x" wall(s) of the "Prime Tower". This is forcing the inactive extruder to heat up, printing the walls on "Prime Tower" and then cool down, on down every single layer, even if this extruder will only print the last layer of the printed part. 

Please revert to the old "Prime Tower", again, this situation increase a lot the:
1 - print time;
2 - the possibility of contamination;
3 - filament material (now I need 4 walls on single layer, rather than 2);
4 - power consuption 

There's no advantage at all on this last "Prime Tower".
Thanks

Cura 3.5.1
[image]

Cura 3.4.1
[image]

[bug_2.zip](https://github.com/Ultimaker/Cura/files/2502128/bug_2.zip)

FreedonNadd, 2018-10-24:
Yes please make it at least as an option.

There is no benefit at all at a normal multicolorprint with single material. It uses lot more of everything with no gain.

diegopradogesto, 2018-10-25:
This new concept of prime tower was an internal requirement and it was created because material adhesion. There were issues with the prime tower falling because adhesion between PLA and PVA (for instance) is not always good and so the prime tower can break at the middle of the print. We know that the time increases but the stability of the tower is better.

Printing with both PLA is less noticeable because the adhesion is good and maybe the previous prime tower is a better option. However we decided not to keep both options because then the code becomes less maintainable.

We won't revert this, sorry.

closed 2018-10-25 by diegopradogesto

diegopradogesto, 2018-10-25:
Oh, that's because we have a setting called `material_adhesion_tendency` that we use to decide which material goes in the outside of the prime tower. If you are printing with the same material, then the value is the same and then the extruder is chosen arbitrarily.

Discussing together with @Ghostkeeper we decided to make this selection stable and select the extruder in order, so the first extruder should be selected.

amagr0, 2018-10-28:
Another tip to the current "Prime Tower" - https://www.youtube.com/watch?v=J86c_7i8jKk

FreedonNadd, 2018-10-29:
@diegopradogesto thank you for your answer. i have an issue with the primetower in regards of the pylon model. https://www.thingiverse.com/thing:21773
one part has always a primetower with both materials/colors. that seems not right to me. But thats maybe a different feature that is not in cura. i dont know. 
thank you for your time.

miriad, 2018-12-19:
@diegopradogesto, any possible way we could reopen this? The oozing and print time this change causes is a huge difference from the 3.4 days, but 3.4 won't work for me anymore (Linux, and the SSL libraries are broken), so I will have to use a different slicer (Probably Slic3r, but I may pay for S3D soon), just because of this issue.

marcottt, 2019-03-07:
Also for me need to change something. Is impossible to make dual color objects without dirty materials

Ghostkeeper, 2019-03-07:
If we can find a way to put a more desirable behaviour in the current architecture, that would have a chance.

But we won't revert to the old prime tower because we've found the new prime tower to be much more reliable with tall prints, and we won't keep both code bases due to maintenance effort.

miriad, 2019-03-07:
Would it be possible to have a switch that makes the "outside" (solid) portion of the prime tower the primary (most used) material for that layer? We can still keep the inside of the tower as a dump location for the second (or more) materials when those materials get used in a layer, and if the other materials arn't used in a layer, they don't need to be dumped into the tower.

I know this won't work well for when printing with PVA as the secondary material, but when printing multiple colors with the same material (PLA, ABS), it should prevent all the oozing problems I've been seeing by only heating nozzles when they're used on a layer.

Ghostkeeper, 2019-03-08:
We currently already make a conscious choice of which material gets the outside tower: It's the material that has the best layer bonding. For instance, if you put CPE in one extruder and PLA in the other, the PLA will always get the outer shell.

miriad, 2019-03-08:
When printing with the same material ( pla  and pla ), is it possible to have the outer material selected per-layer, so it's always the "most used" (or first used, that would work too), rather than always the same?