Ultimaker/Cura #4632: 3.5.1 - Overlapping issue

amagr0, 2018-10-22:
**Application Version** - CURA 3.4.1 / 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Open Cura Project; analyse the toolpath - 1st layer.

**Actual Results** - There's a strange overlap situation in the initial layer.
[image]

[image]

And is also strange the way toolpath is generated. 
From the left to center, it prints on this way.
[image]

And then, from the center to the right it prints on a completely different way.
[image]

Since this section of the print is symmetrical, the expected result should be the same.

There's also other sittuations on the same gcode:
[image]
[image]

smartavionics, 2018-10-22:
Yes, I think they know about this breakage.

Ghostkeeper, 2018-12-31:
The case of it behaving not symmetrically is because the lines aren't aligned symmetrically. It's sort of expected behaviour. Your piece may be symmetrical over the Y axis, but the lines aren't; they are shifted.

The blob in the corner has since been fixed.

The overlapping situation still exists even after our recent refactors in this case. It happens because it finds two short lines very close to each other that can be merged together. It doesn't do any form of detection of whether there is a line in between (as that would be very expensive).

I don't think we are able to fix this elegantly. It happens in few enough cases and with small enough consequences that I think we should accept the overlaps as-is.

closed 2018-12-31 by Ghostkeeper