Ultimaker/Cura #4633: 3.5.1 - Reload STL not working anymore

amagr0, 2018-10-22:
**Application Version** - CURA 3.4.1 / 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - install 3.5.1; open a stl; modify it in CAD, save it.

**Actual Results** - The STL reload feature introduced on 3.5 is not working anymore. the window with the warning does not pops up.

smartavionics, 2018-10-22:
Actually, I have the opposite problem! When I modify the STL, Cura pops up many notifications saying the file has changed and do I want it reload it. I thought they were going to fix this bug months ago, obviously not.

diegopradogesto, 2018-10-25:
This feature was removed in Windows because it was causing errors and crashing because of the behavior with file in USB sticks (https://github.com/Ultimaker/Cura/issues/4495) or in cloud based drives (https://github.com/Ultimaker/Cura/issues/4502).

In Linux it should work properly and not showing the message several times. I thought it was fixed, but since I use Windows, I don't see the messages. However @LipuFei told me that he is also getting those messages many times. Probably a regression...

smartavionics, 2018-10-25:
Multiple file change notifications on Linux have been occurring for many months now. It was never fixed.

diegopradogesto, 2018-10-31:
@Ghostkeeper or @alekseisasin can you confirm if the multiple popups still occur in Linux? I remember to ask you and I remember you were testing it again.

Regarding the original issue, the decision was to remove the feature in Windows since it was causing other issues, (afaik) caused by certain bug in our framework. I'm closing this issue since it's then not considered as a bug. We will test in Linux if it still happens, in that case we can reopen the initial issue.

closed 2018-10-31 by diegopradogesto

lokster, 2018-11-13:
I can confirm the multiple reload messages issue on linux with Cura 3.5.1.
In fact, it's quite annoying - when I export STL from Blender, I have to click 20-30 times (or MORE!) in order to close all popup messages. Initially it looks as if nothing is happening, because the identical messages are too many, and Cura displays only a limited number of them.
An easy workaround will be to add "Close All" function :)
OR just don't display a message about file X being changed, if there is already a message about this file on screen.