Ultimaker/Cura #4634: 3.4.1/3.5.1 - Optimize Wall Printing Order issue

amagr0, 2018-10-22:
**Application Version** - CURA 3.4.1 / 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Open Cura Project; analyse the toolpath.

**Actual Results** - The outer wall of the part seems not to be taking advantage of the "Optimize Wall Printing Order" feature.

[image]

[image]

[bug_7.zip](https://github.com/Ultimaker/Cura/files/2502296/bug_7.zip)

smartavionics, 2018-10-22:
I agree that in this situation it would be nice if the outer walls were printed in sequence from the outermost to the innermost but the reason they are not is because of the following constraints:

1 - to ensure that the outer wall does not get flow reduced when it lies very close to a hole in the part, the outer wall is printed before all of the holes' outer walls.

2 - the holes' outer walls are printed before any of the inner walls.

So it ends up that the two walls that are inside the part's outer wall get printed last. 

It would be possible to detect when it is safe to print all of part's outer walls before any of the walls that surround the holes but, IMHO, the optimization code is already complicated enough and the benefit is so little that it's not worth doing.

amagr0, 2018-10-25:
Hello @smartavionics, what's the difference in this case?
[image]

smartavionics, 2018-10-25:
OK, that looks to me that you are not specifying outer wall first in that example. In your first example you were printing the outer wall first.

smartavionics, 2018-10-25:
Just noticed something a little weird, the spacing between the walls is different. What line widths are you using for outer and inner walls?

amagr0, 2018-10-25:
Yes, the outer wall order is different on the two images. My question is, why in the second image the perimeters of the exterior wall are all printed sequential, and on the first image not?

smartavionics, 2018-10-25:
When the outer wall is not printed first, it is OK to print from inner to outer as an unbroken sequence which is what it is doing in your second example. When the outer wall is to be printed first (as in your first example) it can't always print all of the inner walls that lie next to the part's outer shape because some of those walls could also go around a hole inside the part so, to be safe, it first outputs the outer wall of the whole part it then goes off and does the outer to inner walls for the holes and then it comes back and finishes with the inner walls of the whole part. As I mentioned above, the code could be changed to detect when it would be OK to print all of the walls around the outline of the part in an unbroken sequence but I don't feel it's worth the effort and complexity.

amagr0, 2018-10-25:
OK, tks for the explanation.