Ultimaker/Cura #4637: "[3.5.1] Multiple issues when displaying the Cura application interface."

bliser3d, 2018-10-22:
**Application Version** - 3.5.1
**Platform** - ALT Linux, KDE5, Plasma Color Schemes - "Breeze Dark variant" end "Breeze High contrast".
**Printer** - Any

**Steps to Reproduce** end **Actual Results**
Red color highlights some (not all) problems.
1. Cura themes - **cura-dark**, KDE5 Plasma Color Schemes - "Breeze Dark variant" end "Breeze High contrast" (I use it all the time.)
[image]
For comparison, version 3.4.1
[image]

2. Cura themes - cura-dark, KDE5 Plasma Color Schemes - "Breeze Dark .....".
[image]

[image]

[image]

3. Cura themes - **cura-light**, KDE5 Plasma Color Schemes - "Breeze Dark .....".
[image]

Further I will not continue - there are still a lot of shortcomings in the interface.

**Expected results**
1. When using the light theme in the Cura, all the interface elements of the application must match the settings for the light theme.
2. When using the dark theme in Cura, all the interface elements of the application must match the settings for the dark theme.
3. The display of all interface elements of the Cura application should not depend on the color theme used in the KDE.

**Additional Information**
Similar issues were reported in: https://github.com/Ultimaker/Cura/issues/4522

I hope I did not tire you.

bliser3d, 2018-11-14:
This problem is completely repeated in version 3.6.0.

bliser3d, 2018-12-28:
This problem is completely repeated in version 4.0 (beta).
Maybe someone knows when this problem can be eliminated?

matibarriento, 2019-01-16:
Yes, I have the same issue with KDE and Cura version +3.6

captNtinker, 2019-03-18:
[image]
Issue with my screen being stuck to the right side (minimized) and when large, can't see the menus open?

captNtinker, 2019-03-18:
can't drag it off, it's stuck to the right side :(

fieldOfView, 2019-03-18:
@captNtinker that is a different problem altogether, but I guess you already came to the same conclusion since you opened a new issue here: https://github.com/Ultimaker/Cura/issues/5443

captNtinker, 2019-03-18:
Yah- I wasn’t sure how to put in the question- it’s got to be a window issue - still can’t get it to work- gonna ask the tech at work today see what he says! Might get a solution?

Ghostkeeper, 2019-03-18:
I'm unable to reproduce this problem, not on Ubuntu 18.10 (using theme Yaru-dark) nor on Fedora (I don't know which theme my colleague was using, but it was dark).

bliser3d, 2019-03-19:
Thanks **yodor** (https://github.com/Ultimaker/Cura/issues/4826) for the prompt. 
Used the command "QT_QUICK_CONTROLS_STYLE=Desktop /home/User/Cura/AppRun".
It helped me.

[image]

[image]

Small remarks remained, but this does not prevent the Cura from using.
Thank you all. I close the topic.

closed 2019-03-19 by bliser3d

muesli, 2019-03-23:
@bliser3d That's a work-around at best. This is still broken in 3.6 and 4.0.

bliser3d, 2019-03-24:
Some shortcomings in the dark theme, I tried to fix it in #5499.

Ghostkeeper, 2019-03-26:
We're planning to replace the system-themed parts in the preferences dialogue for Cura-themed elements, which would solve this issue. Combining the two themes is never a good idea.

VCF, 2019-04-20:
@bliser3d , could you please clarify how you passed `QT_QUICK_CONTROLS_STYLE=Desktop /home/User/Cura/AppRun` to the application? Is there a command line flag I should use in conjunction with it? Also, I presume the command is a string literal, as opposed to needing to substitute for a specific file path?

Running under Mint 18.3, a very large number of the menus and text entry boxes are white-on-white, such that I have to interact with the system by trial-and-error. Thanks!

bliser3d, 2019-04-23:
**VCF**, 
**QT_QUICK_CONTROLS_STYLE=Desktop /home/User/Cura/AppRun** - this is the start command **Cura**.
If interested, you can improve the interface further and in this way - #5499

e-anima, 2019-05-17:
bliser3d thank you. Totally fixed it for me.

captNtinker, 2019-05-17:
The resolution for me was to unload it and load an earlier version :( but it’s working again!

captNtinker, 2019-05-17:
Rather the old version is working fine- I would have to hit Maximize just to get to it and then hit escape to even work in it, then couldn’t anchor it out of the way- it went way off the screen where I couldn’t get it :/

bliser3d, 2019-05-18:
In version 4.1-beta, problems with the display of the interface (previously indicates by me) remained.
Fixed only the display of the window "Machine Settings" - thanks for the correction.

Ghostkeeper, 2019-05-20:
In 4.1, only the machine settings and the add printer dialogue was replaced with Cura-themed elements. The rest of the preferences menu is still system-themed. It's a lengthy process, sorry.