Ultimaker/Cura #4640: [3.5.1] Mesh Type "Modify settings for overlap with other model" produces Unable to slice

sneaky-tricky, 2018-10-23:
Simple overlap of model with "Modify settings for overlap with other model" mesh can not be sliced. Same profile and project works fine in 3.4.1.

[CFFFP_Better_Temperature_Tower_v5_240-160.zip](https://github.com/Ultimaker/Cura/files/2504531/CFFFP_Better_Temperature_Tower_v5_240-160.zip)
[image]

Vandrasc, 2018-10-24:
Hi @sneaky-tricky , the problem lays at the print sequence - one at a time- , so in this case, it is expected behavior that the slicing is not being triggered, because the disallowed areas are overlapping. 'The one at a time is only possible if all models are separated in such a way that the whole print head can move between the models.'

Can you please confirm if Cura triggers a slice when Print sequence is set to 'All at Once'? 

Thanks!

sneaky-tricky, 2018-10-24:
Oh, you're right. I didn't realize that "Modify settings for overlap with other model" mesh doesn't work in "One at a Time" mode. Everything's fine in "All at Once" mode.

closed 2018-10-24 by sneaky-tricky

Ghostkeeper, 2019-01-02:
See also #4193.