Ultimaker/Cura #4648: 3.5.1 - Shortcuts left panel not working.

amagr0, 2018-10-25:
**Application Version** - CURA 3.5.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - install 3.5.1; open a stl; try the shortcuts from left panel.

**Actual Results** - The shortcuts from the left panel are not working. 

[image]

rburema, 2018-10-26:
Hi @amagr0, Did you select the model first? I haven't been able to reproduce this yet.

amagr0, 2018-10-26:
If you not select the model the panel stay inactive, so yes, the model was selected.
(did you know well Cura software?)

[image]

rburema, 2018-10-26:
My apologies, I though you might have selected the model afterwards, to show us which shortcuts you meant. In retrospect, it didn't make that much sense for me to assume that, sorry.

alekseisasin, 2018-10-31:
I close the issue because it is duplicated: https://github.com/Ultimaker/Cura/issues/2556

closed 2018-10-31 by alekseisasin