Ultimaker/Cura #4654: Feature request - Prime Tower dynamic position

amagr0, 2018-10-25:
Could be nice to not having to worry about the position of the Prime Tower. Positioning the Prime Tower to far way from the model could result on some ooze material, and putting near the model will require a repositionoing of the Tower everytime the next part to print is bigger that the previous. So a dynamic way to controle this could be by defining an offset from the model.

This feature could have two settings:
1 - offset from the printed part, in mm.
2 - position on the build plate (upper left/center/right, middle left/center/right,  lower left/center/right) 

[image]

alekseisasin, 2018-10-30:
I think the best solution could be user can select any position on the build plate. 
Like "Default" and "Dynamic" where a user defines the X and Y coordinate.

fieldOfView, 2018-10-30:
The prime tower position is already settable; see the Prime Tower X Position and Prime Tower Y Position settings.

alekseisasin, 2018-10-31:
Right,
@amagr0 , I think you can easily adjust the prime tower position on the build plate. 
Dynamically setting the prime tower offset from the model is a quite specific case and will not add value for most of the cases. Maybe in future. If you can explain why it still needs to be added then we can consider your examples. 
For now, I suggest close the issue.

amagr0, 2018-10-31:
Q: "If you can explain why"
-
A: Positioning the Prime Tower to far way from the model could result on some ooze material, and putting near the model will require a repositionoing of the Tower everytime the next part to print is bigger that the previous.

Cura needs to start to be more automated with this "simple" things. 
Ultimaker is now making machines for the professional/industry segment, the type of user for this segment should not be worried about this things, so a dynamic way to control this should be realy helpfully.

I don't know the complexity to develop this type of feature, but I think if Ultimaker is dealing with professional/industry segment, needs also to deliver professional solutions.

AbeFM, 2019-04-05:
I tend to agree, picking the angle and distance seems pretty easy, and automatically moving it to a new corner if that corner is full doesn't seem too bad.

I like the argument about consistent distance.

I keep having issues where changing a setting will move the position to an invalid one, then I need to enable the tower, set a position, and disable it. Implementing this feature wouldn't be the fastest way to solve not letting a bad position on an unused tower from preventing slicing, but it would kill two birds with one stone.

Ghostkeeper, 2019-04-11:
Because polygon offsets in Python are pretty tough, picking an angle and distance analytically is hard. And it'd need to be implemented in our Python front-end since it needs to be shown to the user before slicing. We would probably solve this by using the "arrange" functionality as if placing a square or circle on the build plate.

AbeFM, 2019-04-25:
I would certainly be ok trying the pillar in any of 100 spots, starting close to the model's center, and if there's a conflict pick the next one. I assume the arrange feature works that way?

Ghostkeeper, 2019-04-29:
Yeah the arranger creates a render of the build plate using 2mm pixels, indicating whether something is inside that pixel or not. And then it finds the position to place a model that is closest to the centre by trying out positions in that pixel grid.

AbeFM, 2019-11-05:
Is that then something which could be done? Bumping the tower around is, when things are going smoothly, one of the bigger time sinks in just banging out a print.

Ghostkeeper, 2019-11-07:
Yes. Well, of course it doesn't jive with the current X/Y position settings so we'd need an additional setting then.

This can be done, but it doesn't have enough priority for us. From our statistics we know that the prime tower position is rarely modified in dual extrusion prints. By far the majority of prints are way too small that the prime tower position is an issue. We'll defer this for later. I think it's a nice quality-of-life improvement, but for most users it would be better to spend time on different things.

closed 2023-07-13 by github-actions[bot]