Ultimaker/Cura #4657: [3.5] Little misplacement of a tree support circle

Liger0, 2018-10-26:
**Application Version**
3.5.0

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use tree supports

**Actual Results**
There is a little misplacement on a support layer. You can see the inside, so it will have an empty area thus making the support weak.

**Expected results**
The supports should be linear. 

**Additional Information**
[image]
[CCR10S tree support.zip](https://github.com/Ultimaker/Cura/files/2517214/CCR10S.tree.support.zip)

alekseisasin, 2018-10-31:
Hi @Liger0, Thank you for your report. We know about this edge case and it is expected behaviour with the current implementation. In a short-term, we will not be able to fix it, sorry. What you can do, try to adjust the Tree Support Collision Resolution. (please check related settings to Tree Support), I hope it will improve quality.

Liger0, 2018-12-08:
> Hi @Liger0, Thank you for your report. We know about this edge case and it is expected behaviour with the current implementation. In a short-term, we will not be able to fix it, sorry. What you can [...]

So now I am facing related but more extreme situations.
In some cases, a support layer is completely skipped, in some other case, some layer is completely printing in the air and so the tree won't be printed and the print will fail.
I'll attach a screen and the model.

[tree.zip](https://github.com/Ultimaker/Cura/files/2659251/tree.zip)
[image]

Ghostkeeper, 2019-12-23:
This problem should be fixed in Cura 4.1.

closed 2019-12-23 by Ghostkeeper