Ultimaker/Cura #4658: [3.5] Random useless supports

Liger0, 2018-10-26:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Use supports

**Actual Results**
There are random supports which are not required

**Expected results**
THhere shouldn't be any wasted support

**Additional Information**
[image]
[image]

[CCR10S wasted support3mf.zip](https://github.com/Ultimaker/Cura/files/2517310/CCR10S.wasted.support3mf.zip)

rburema, 2018-10-26:
Hi @Liger, Very likely a duplicate of #4606 and #4653 , please continue any discussion over there (the first one that is). cheers, Remco.

Ghostkeeper, 2018-10-29:
I can't reproduce the issue in this picture with your model. It's probably dependent on the exact positioning of the model.

Reinforcing that argument is that your model has a lot of extra geometry on the inside, which can confuse Cura as to which parts of your volume are inside and which side are outside.

[image]

closed 2018-10-29 by Ghostkeeper