Ultimaker/Cura #4659: Timer in QualityManager.py introduces non-determinism?

fusentasticus, 2018-10-26:
There's a timer in code that's run when Cura profiles are loaded:
https://github.com/Ultimaker/Cura/blob/83f0213aa529491cc83ed3f1868ae532d3b59f45/cura/Machines/QualityManager.py#L63

Is that related to https://github.com/Ultimaker/Cura/issues/4162, see my pained observations at end of the thread? (And various user complaints?)

A timer in initialization code of data structures as complicated as the ones parsed from file names and INI stuff is hazardous. It may easily introduce non-determinism in whatever the meaning of all this profile stuff have evolved into. I'd suggest that it should be eliminated unless it clearly doesn't mess anything up --- especially if it bumps the initialization time of Cura up the polynomial hierarchy at the same time !!

nallath, 2019-01-07:
It actually decreases the loading time. It basicly is there to ensure that if 50 changes happen right after oneother, that the entire model doesn't get updated (and re-calculated 50 times). By putting a relatively short timer (that get started every time a change occurs), we only get a single re-calculation of the model.

closed 2019-01-07 by nallath

fusentasticus, 2019-01-07:
Thanks for looking into it --- I'd still suggest to use hash trees or ledgers instead of timers to avoid recalculations in recursions: just use a table of calls and their results, hash as necessary.