Ultimaker/Cura #4660: Cura 3.5.1 Under Extruding

pinchies, 2018-10-26:
In 3.5.1 the machine filament diameter is not respected, unless a material is selected. In 3.5.1 there is a bug where no material is selected by default. This results in custom printers with 1.75mm filament diameters experiencing severe under extrusion. Verified this issue is not present in 3.5. Verified this issue does occur after resetting Cura.

**Application Version**
Cura 3.5.1 or higher

**Platform**
Mac OS - High Sierra 10.13.6 (17G65)

**Printer**
Any

**Steps to Reproduce**

1. Add a new custom printer
2. Under machine settings, set filament diameter to 1.75
3. Add stl file, and export gcode - save as "test_unknown_material.gcode".
4. Under materials, you will see the material currently selected is "unknown material". Even though in the slicer interface PLA is shown. (see image at bottom of issue)
5. Select generic PLA from the matieral manager, and activate that.
6. Slice again, and save as "test_generic_PLA.gcode".
7. Open both files - observe that the filament length for the unknown material is half as long - it is slicing as 2.85mm filament even though we specifically set 1.75mm in the machine settings.

**Actual Results**
Unless a material is specifically selected, the machine filament diameter is ignored, and the material is sliced as 2.85mm.

**Expected results**
The filament diameter set in machine settings should be used to slice the material.

**Additional Information**
Verified bug after removing and reconfiguring all Cura files.
[image]

rburema, 2018-10-26:
Hi @pinchies, This seems like a duplicate of https://github.com/Ultimaker/Cura/issues/4582 , and was fixed in master very recently. cheers, Remco.

pinchies, 2018-10-26:
I'm satisfied that it looks like it has been fixed. I hope the 3.6 release comes out pretty soon... I lost many hours to troubleshooting this one, first thinking that my printer was on the fritz... this bug will affect every single user who has 1.75 who sets up a custom printer.

pinchies, 2018-10-29:
Hi team, any ETA for 3.6? I've been helping people all day on our facebook group and forum - many people are running into this bug and suffering underextrusion. It's causing people to waste a whole bunch of time diagnosing hardware, when the problem is software.

smartavionics, 2018-10-29:
Hello @pinchies, you could try one of my master branch builds. The most recent is likely to contain the fix although I can't be 100% sure. You can find them at:

https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

pinchies, 2018-10-29:
Great, thank you, I'll try that out.

diegopradogesto, 2018-10-31:
3.6 BETA is live from today. You can try if this is fixed for you.

diegopradogesto, 2018-10-31:
I'm closing this issue since it's considered as a duplicate and because it is fixed in our 3.6 branch and master.

closed 2018-10-31 by diegopradogesto

Picklwarrior, 2020-11-14:
I am literally having this issue in 4.8. I didn't upgrade from 2.7 for years because ages ago I **finally** figured out this was my problem and I refused to upgrade until I was sure it would never be an issue again. I finally upgraded because I really needed the support remover function and now I'm having the exact same problem.

Ghostkeeper, 2020-11-17:
You should not be able to get any 2.85mm materials to load in a printer that uses 1.75mm. Do you have reproduction steps for us?