Ultimaker/Cura #4661: Cura 3.5.1 slicing bug, extrusion outside of the model

Roblue, 2018-10-26:
**Application Version**

3.5.1

**Platform**

Windows 10

**Printer**

Geeetech Clone, Prusa i3 Mk 2

**Steps to Reproduce**

Select Cross 3D as infill pattern, 20% infill density, connect infill lines on, connect polygons on

**Actual Results**

Cura wants to print random lines marked as "Infill" in the layer view outside of the actual print. It happened with several models.
[image] (Color scheme: Feedrate)

[image] (Color scheme: Line Type)

**Additional Information**

I wasn't able to reproduce the problem with any other infill density than 20%. The problem is not affected by any settings other then the ones named (as far as I tested).

Additional pictures and the poject file:
[SlicingBug.zip](https://github.com/Ultimaker/Cura/files/2518201/SlicingBug.zip)

smartavionics, 2018-10-26:
Hi, I'm pretty sure this has been fixed for the next release.

See https://github.com/Ultimaker/Cura/issues/4483.

rburema, 2018-10-26:
Hi @Roblue,
I've tested this in 3.5.1 (bug does occur for your model) and a recent master-build (fixed, for this model as well). So I agree with @smartavionics, this is a duplicate of that issue, and will be fixed in the next release.

P.S, As an aside; 'save' instead of 'export' causes settings to be saved as well, this allows for quicker set-up on our end.
cheers, Remco.

Roblue, 2018-10-26:
Thank you for your quick answers.

I may have overlooked the other thread while searching for the issue.
Thanks for the hint @rburema!

Best regards, Rob

diegopradogesto, 2018-10-31:
I'll close this issue since it is a duplicate. Also you can check out Cura 3.6 BETA that is live from today.

closed 2018-10-31 by diegopradogesto