Ultimaker/Cura #4665: Anet A6 stops at initial G-code and waiting for user confirmation

sim31r, 2018-10-26:
Init G-code, standart for Marlin, I added only a pause of 5 seconds (for table check level)
error occurs 50% of the time, not stable
At old version 3.0 everything was fine 
At new version 3.5.1, 3.5.0 this error

Ibitial G-Code, standart and my small parts

> G21 ;metric values
> G90 ;absolute positioning
> M82 ;set extruder to absolute mode
> ;M107 ;start with the fan off
> G28 X0 Y0 ;move X/Y to min endstops
> G28 Z0 ;move Z to min endstops
> G1 Z20 F9000 ;move the platform down 15mm
> G92 E0 ;zero the extruded length
> ;G1 F200 E3 ;extrude 3mm of feed stock
> ;G92 E0 ;zero the extruded length again
> G1 F9000
> G1 X122 Y122
> M105
> M300 S5000 P250
> G1 Z0
> M105
> M84 ;steppers off
> M105
> ;G4 S5
> M0 S5; wait 5 second ot less at zero level for paper test sometimes
> M105
> M117 Printing.... 

At this code, 3D printer stops (around M84 command). The printer stops forever, until you press the button on the front panel. Whith the message "Waiting for user..."
But I am not send "M0" - endless waiting
M105 added for stability, but it is not necessary
The error is big. The printer is waiting with a heated nozzle and a table (If, for example, went to drink coffee and forgot about the printer)
If export the G-code to Repeater Host, work fine, pause 5 seconds if specified. 
I think the Cura 3.5.1 (Cura 3.5.0) sends the wrong code to the printer, in the case when there is a long wait (heat, pause).
I can see the exchange in the COM port, I think there is a reason.

sim31r, 2018-10-26:
I try the beta version. 
Ultimaker Cura-3.6.99-fix_press_to_resume-win64.exe
At the moment, once launched, all is well.

sim31r, 2018-10-28:
At the moment everything is working well with v3.6.99. Printed 10-15 small objects.

diegopradogesto, 2018-11-08:
This issue seems to be fixed in 3.6 BETA. Next week we will release the stable version. Thanks for the tests!

closed 2018-11-08 by diegopradogesto

BatLab1, 2018-11-14:
I have version 3.6.0 and I have the same problem, totally ruined hours of printing. So what is a good for this issue.

HuguesDug, 2018-12-20:
Same issue.

Anet A8, marlin firmware 1.1.9
Cues 3.6

After heating, request user confirmation randomly.

But is still there.