Ultimaker/Cura #4671: [3.5.1] File menu Save/Export options should be more verbose

smartavionics, 2018-10-26:
**Application Version**

3.5.1

**Platform**

Any

**Printer**

Any

**Steps to Reproduce**

Click on File menu.

**Actual Results**

File menu opens but it's not obvious what the Save and Export (and Export Selection) menu items actually do.

**Expected results**

File menu should show sufficient information so that user knows which item to select.

**Additional Information**

The "Save" item should be "Save Project" (if that's what it does).

The "Export" item should be "Export Models" (if that's what it does).

Ditto for "Export Selection".

diegopradogesto, 2018-10-31:
We recently changed the behaviour to be as it is now. We consider that project files are the main type of files in Cura (the ones that contains the whole scenario to load the exact same status) so that's why we use the "Save" button.

Then the "Export" is used as in some many other software to export what you have in the scene as a different format. Not always is a model, but it can be also the upf, or gcode or some other format. 

"Export selection" stands for the same operation as before, but only applies to the object that you have selected. 

Having said that, I really understand better how it is now, not how it was before. It's my opinion. 

Maybe @LukeChen-Ultimaker has better arguments why this was decided.

LukeChen-Ultimaker, 2018-11-02:
1. If "**Save Project**" brings more clarity to you as a user, then let's change it and see how people react to it.
2. Export does not only mean " export model" to another file format. You can in fact export to Gcode which contains more info than the model itself. So I would not recommend to use "Export Models".

fieldOfView, 2018-11-02:
> Export does not only mean " export model" to another file format. You can in fact export to Gcode which contains more info than the model itself. So I would not recommend to use "Export Models".

I would argue that exporting gcode and exporting models should be split into two menuitems. They are fundamentally different things for the user. Exporting gcode is the core purpose of the application (for many users), while exporting models as a certain file format is just a byproduct of being able to load and save models (and mostly only a usecase for a few users who don't have better tools to convert models).

* Save Project
* Save Sliced data
* Export models

Note that I'm suggesting "Save Sliced data" as opposed to "Save GCode", because the sliced data may actually be saved as UFP, or another format.

Ghostkeeper, 2019-02-19:
We've decided from a UX perspective that we like the current situation pretty well and the proposed situation equally well (opinions are divided). But combined with the amount of work estimated on this, that turns the decision into a no. We're not going to do this. Sorry.

closed 2019-02-19 by Ghostkeeper