Ultimaker/Cura #4674: X-Y Overrides Z only considers ceilings

AbeFM, 2018-10-27:
**Application Version**
3.5.1

**Platform**
win10-64

**Printer**
Homebrew

**Steps to Reproduce**
Slice with supports. Vary z-overrides-xy with support XY and Z distances set.

**Actual Results**
[image]
You can see in the first image, there is missing support carved out to respect XY over Z

In the second picture, you can see Z prioritized.

Looking under the support, you can see XY is always respected.
https://community.ultimaker.com/topic/25049-support-floors-not-respecting-z-overides-xy/?tab=comments#comment-222884

**Expected results**
That the support floor should hug the part as the roof is set to do, or give it separate settings.

**Additional Information**
The current behavior can lead to failed supports as they may print in mid air.

AbeFM, 2018-12-13:
**Application Version**
3.6.0

**Platform**
win10-64

**Printer**
Dual Extruder Marlin 1.9

**Steps to Reproduce**
Slice with supports. Vary z-overrides-xy with support XY and Z distances set.
New info: Still happening in 3.6.0.
I wanted to add I've had minimum x-y distance set to infinitesimal values and the top behaves as expected but the floor does not.

_Am I the only person seeing this?_

It's making printing with PVA challenging. I've started from fresh configs.

[image]

With all distances set to 0 you get support that touches but lose the ability to keep vertical supports off  your part.

AbeFM, 2018-12-13:
Can someone add a "bug" label to this??

AbeFM, 2018-12-18:
**I DON'T WANT TO BE AN ASSHOLE HERE... CAN SOMEONE _PLEASE_ tell me if they see this?!?!?!**

I just want to know if it's real. Is it a bug? Am I doing something wrong?

*CAN YOU TELL ME WHY NO ONE WILL ANSWER ANY OF MY 3 THREADS ON THE FORUMS AND HERE?!?!?!*

I haven't broken some rule, and I don't imagine  you INTENTIONALLY make prints fail, so please tell me what do to do get ANY feedback from ANY person.

fieldOfView, 2018-12-18:
I can see this. I just don't have anything constructive to add.

AbeFM, 2018-12-18:
That is super constructive. For months I didn't even know if I was just doing something wrong.

Thank you *truly* for letting me know. If no one cares, ok, but at least I want to make sure it's real.

fieldOfView, 2018-12-18:
All I can say is that the team have been fairly busy implementing the new UI (see the Cura 4 beta released today)

AbeFM, 2018-12-18:
It's not important that I  brought this up before 3.5.1 hit the "shelves" - but I would appreciate something like a tag calling it an "issue" or something. If it's real, and there's no time, ok fine.

But I get the feeling it's ignored. I don't wish to set priorities for the dev team, but I would like to know that they are AWARE of the issue and have it somewhere on their priority list.

I personally think it's pretty important - it literally prevents me from printing with Cura for my most advanced projects where it used to be my go-to slicer. I still use it for simple stuff.

I'm happy to answer questions and run tests, but my goal is just to get it as a recognized issue.

AbeFM, 2019-04-04:
Checking in on this one. The new version doesn't seem to have fixed this.

[image]

I've seen people demonstrate this working correctly, but since the low 3.x versions I have just not had supports which stick in a long time - either I space them off the model and they don't stick or I butt them up to the model and ruin the surface finish by having supports embedded into the wall.

The existence of the settings implies this is something everyone wants to fix, but it doesn't work.

I've tried turning support expansion on and off, changing vertical distances, angle, connecting supports, stair step height, join distance, interfaces on and off..... While I have seen posts where people have made it work on the main forums, I cannot repeat it.

This has been going on for the better part of a year, and it makes CURA unusable. Once again. I am willing to run any tests.... 

Could someone comment on the status of this, any ideas on what I could check out?

Ghostkeeper, 2019-04-05:
Hold on, was this about the case where X/Y overrides Z or where Z overrides X/Y?

Ghostkeeper, 2019-04-05:
It seems indeed that stair stepping is interfering with something there. The stair stepping itself seems to be broken as I can't get it to create any steps any more.

AbeFM, 2019-04-05:
> It seems indeed that stair stepping is interfering with something there. The stair stepping itself seems to be broken as I can't get it to create any steps any more.

That's interesting. I wasn't sure what it did, and I saw no change when I did it, I just ignored that as I don't use it (getting soluble supports to break away easily is rarely a motivation for me).

To answer: These are for "z overrides x-y".  I tried it both ways, but xy overrides z didn't work any better. :-)
More usefully:  everything works as it should on the UPPER end of the supports, my point/report is that the same rules are ignored on the support floors.
This leads to supports not sticking very well to the model, and failed prints.

Ghostkeeper, 2019-04-11:
Yeah the stair stepping is only applied to the bottom of the support where it rests on the model. It's intended to create sort of a staircase effect on the bottom of support where a slowly sloping surface (like in your example) would not have a slowly sloping support bottom side but a bottom side that goes up in bigger steps. This reduces the contact area on the bottom without making support much weaker.

AbeFM, 2019-04-25:
Anything I can do here to help?

Everytime I think "I should give up and just buy an UM" for this project, I come back to "my failures are from the slicing, not the machine".

The only realistic solution I see to making these supports work is setting the support angle and expansion so high that your object is encased in solid PVA. Even so, I have issues getting a clean outer surface once I let the X-Y distance get much below half of the line width.

Ghostkeeper, 2019-04-29:
No indeed, this problem is not specific to any printer. Switching to an Ultimaker won't solve it for you.

We have this issue in our backlog right now. I think it's pretty critical but there are so many critical issues that I can't promise that it'll be fixed for the next release.

AbeFM, 2019-05-05:
That's awesome. My problem isn't necessarily worse than anyone else's -
knowing it's on the list is great.

Anything I can do to help I'm happy to do!

AbeFM, 2019-08-11:
> No indeed, this problem is not specific to any printer. Switching to an Ultimaker won't solve it for you.
> 
> We have this issue in our backlog right now. I think it's pretty critical but there are so many critical issues that I can't promise that it'll be fixed for the next release.

You're swamped even worse now, I get that.... But even at only bringing it up once or twice a year it's starting to sound old.

I really CANNOT print without my supports sticking to the print. If there's ANYTHING I can do to help, let me know.

Vandrasc, 2020-06-22:
Hi @AbeFM , we applied a fix in our master build and we think it may also have resolved the issue you described. Sorry for the long reply, we did have a lot of work on our backlog. 

I had a look at different models, as I did not have this particular model, and couldn't encounter the issue you described. I hope this fixes the issues on your model as well. 

Thank you

AbeFM, 2020-06-23:
It has been merged in for a little while now, right? Is it in the current
release version?

I'd be happy to load up a troublesome model and slice it.

Ghostkeeper, 2020-07-01:
Let us know if you still experience it with that build, or with the 4.7 beta or later (which is not yet released at the moment of this writing).

closed 2020-07-01 by Ghostkeeper