Ultimaker/Cura #4675: More Crashes in 3.5 (3.5.1)

Michaels-3dPrinting, 2018-10-27:
**Application Version**
3.5.1

**Platform**
Windows 8

**Printer**
None

**Steps to Reproduce**
tried to bring up Cura it crashed and then made me do backup which lost all configurations.

**Actual Results**
Tried loading new settings by hand and now the heated bed does not appear.

**Expected results**
If it had not crashed and wiped out my settings I would expect to just use it as normal

**Additional Information**
 Sorry no screen shots reverting to 3.4 cannot play games with an unstable version.

rburema, 2018-10-28:
Hi @Michaels-3dPrinting, Can you share the log at least? cheers, Remco.

Michaels-3dPrinting, 2018-11-05:
Last log before reverting to 3.4 - which is working fine.

rburema, 2018-11-05:
Did you attach any file (cura.log or any of the others) to that last comment? I can't see any.
We need to have at least some information, if we want to have any hope of fixing it.

Also, depending on the nature of the crash, you might want to try out the 3.6-BETA release: https://community.ultimaker.com/topic/25124-introducing-ultimaker-cura-36-beta/

Michaels-3dPrinting, 2018-11-05:
[cura.log](https://github.com/Ultimaker/Cura/files/2549384/cura.log)
Must have used windows explorer and now changed to Firefox and this seems to be working in uploading

closed 2020-06-19 by nallath