Ultimaker/Cura #4676: Experimental

53peterp, 2018-10-28:
Version 3.5.1
I've been using the multi plate functionality found in: 
Settings/General - Experimental

When loading multiple objects on the bed the file names appear in the list give easy identification of each object.
The problem is:
When the project is opened at a later time the file names are gone and in their place is object1, object2 etc.

smartavionics, 2018-10-28:
Hi. Yes, it's even pain for single objects on single build plates because it means that when you open a project you can no longer use F5 to reload the model(s) because the model file name is not stored in the project.

Here's a suggestion team: I know you don't want to or can't store the model file names in the project but how about this. Add a file menu option to replace the selected model with a model loaded from a file. The model's position and scaling would not be altered so it should be able to update models in projects that way rather than how it is now which is that the existing model has to be deleted and a new model loaded/moved/scaled.

BagelOrb, 2018-11-04:
Why not store the file name in the project?

smartavionics, 2018-11-04:
I don't know why they don't store the file name in the project but this has been requested before and they haven't indicated that it's going to happen ever. As it is now, it's a pain to update a model in a project which is crazy.

BagelOrb, 2018-11-04:
Ohh so it might just be a priority-thing rather than not wanting to or not being able to...

Ghostkeeper, 2019-02-06:
Storing the filename in the file doesn't solve the problem if you're sharing that file with someone else.

As far as I'm concerned, pressing F5 after loading a project file should reload that project file, not necessarily the files that the project file used to consist of. Reloading files individually is a difficult recursive problem. Reloading the project file means that we need to somehow store the project file name somewhere in the scene but that is more doable.

smartavionics, 2019-02-06:
How about what I say above:
> Here's a suggestion team: I know you don't want to or can't store the model file names in the project but how about this. Add a file menu option to replace the selected model with a model loaded [...]

This would avoid saving the file name in the project but still preserve whatever scaling and positioning had been done previously.

BagelOrb, 2019-02-06:
My 2 cents:
As a user I want my F5 behavior to work even if I close Cura in between.

nallath, 2020-06-19:
Names are now better stored in the project files. So I guess this can be closed.

closed 2020-06-19 by nallath