Ultimaker/Cura #4677: Unable to save printer settings to material profiles.

NickNonsense, 2018-10-28:
**Application Version**
Cura v3.5.1

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Steps to Reproduce**
Material > Manage Material.. > Create / after creating two separate material profiles and back to the UI dash, can't save independent print settings between material profiles.

**Actual Results**
Setting independent print settings between two material profile doesn't save to they respected material profile. Such as print speed, printing temp, retraction etc. Instead, I'm given an option to discard, keep or create a new "Print Profile"

**Expected results**
Option to save print settings to materials profiles, instead of having multiple printer profiles for small tweaks in print settings between different materials like temperature, retractions, speed. 

**Additional Information**

alekseisasin, 2018-11-13:
Hi @NickNonsense , It worked for me in Cura 3.6. If it still does not work for you in Cura 3.6 can you list the reproduce steps again?

NickNonsense, 2018-11-13:
I'm surprised nobody has brought this issue up before but tested in Cura 3.6 and it is behaving exactly the same, I'll take some screenshots as I don't think I've explained the issue clear enough. 

I have my material set here to PLA_Red and you see i have the "Default Printing Temperature" set to 235
[image]

Now i select my second material instead which i would like to print at a much lower temp, 200 degrees and i'm greeted by this window that asks if i would like to "Keep" or "Discard" i want to do neither really as i want the "PLA_Red" Material to keep a temperature of 235 and i want the "PLA_Gold" to have a print temperature of just 200 degrees 

The dialog box reads; "You have a customiszed some profile settings." well yes but its material specific, not printer profile specific since i am using the same printer for both materials but they have separate temperatures not only that but i'd like to set different print speeds and so forth. 
[image]

so i have no option but to "Create New Profile" which has me questioning why we have a material tab at all, is it not just rendered useless now because any matrerial specific settings i have i can not have those settings saved in the materials tab, but i need to create a new print profile instead. I see no reason to use the materials tab apart for data entry for weight, cost etc. 

Have i overlooked how the layout is set out maybe, is this intentional?

I have seen this tab in "Materials > Print Settings" where i can set various settings here but these settings are not hounered in the main UI window as you can see in the background where the "Default Printing Temperature" is still at 235 degrees, Are those settings here just for reference, where they have no effect on slicing settings?
[image]

nallath, 2020-11-03:
You can do this with the material settings plugin.

closed 2020-11-03 by nallath