Ultimaker/Cura #4680: 3.5.1 won't export custom profiles

Thisismydigitalself, 2018-10-28:
**Application Version**
3.5.1

**Platform**
win10/64

**Printer**

**Steps to Reproduce**
/Manage profiles/Chose custom profile/Export

**Actual Results**
Cura won't export

**Expected results**
export any custom profile with either thier given name or rename them, then save.

**Additional Information**
For me this feature workes in Cura 3.4.1, not in 3.5.1.

Cura 3.4.1 export, you can see the save-as inclues *.curaprofile extension
[image]

Cura 3.5.1 does not.
 
[image]

Vandrasc, 2018-10-29:
Hi @Thisismydigitalself , I am not able to reproduce this, could you please share more information, like steps to reproduce, do you try to export a customized profile or just a clean duplicate? Thank you!

Thisismydigitalself, 2018-10-29:
Steps to reproduce are simple and as described in my initial post above. i tried them all: 
- Creating new custom profile
- Duplicate protected profiles with a new name and then export the newly created custom profile
- Renaming custom profiles
- Activating and disactivating profiles before trying to export
 
Nothing works. i get the same export dialog and with file type missing and even if i disregard this problem and give the file a name and save it, no file is being created.

Vandrasc, 2018-10-29:
Argh, I think I know what happened. @Thisismydigitalself , could you please see if your Cura profile reader (plugin installed by default) is disabled? You can check that in ToolBox>Browse Packages>Installed>Cura Profile Reader. 
After you enable the plugin it should work as expected and the extension option should be visible again. 

Only when disabling the plugin I can reproduce the issue as described by you, is it possible that you, maybe, disabled it in a previous Cura version? 
Hope it works, else I will hear from you :)

Thisismydigitalself, 2018-10-29:
@Vandrasc
You're the man! this dod the trick. Now i wonder why this plug is not selected by default when installing Cura. i can now export and import back profiles.

Thank you.

Vandrasc, 2018-10-30:
"Now i wonder why this plug is not selected by default when installing Cura. i can now export and import back profiles." - That is a mystery.

Glad it works!

closed 2018-10-30 by Vandrasc