Ultimaker/Cura #4682: filament advancement

shellcracker789, 2018-10-29:
Hi
I have a Creality CR-10s that has been modified for two color. I have Cura setup to print 2 color but the one thing i cannot figure out is this. The first time the 2nd color is activated i need the extruder to advance 100mm before starting its doing anything else. Is there a setting I am missing?

alekseisasin, 2018-11-13:
Hi @shellcracker789 , what do you mean "modified for two color. "? Do you mean two extruders? If so when I add Creality CR-10 it has by default only one extruder.
I think you need to change the setting "Enable Prime Blob"

nallath, 2019-03-01:
Closing due to inactivity

closed 2019-03-01 by nallath

Ghostkeeper, 2019-03-01:
You could do either:
* add a big +100 E move to the extruder start g-code, and retract again by 100mm in the extruder end g-code, or
* put the big +100 E move in the global start g-code, and accept that the filament is already in place before the first extruder starts printing. You'll have it in the same place while the first extruder is printing on subsequent layers anyway.