Ultimaker/Cura #4683: [master] layer view missing layer slider

smartavionics, 2018-10-29:
**Application Version**

master

**Platform**

Linux

**Printer**

CR-10

**Steps to Reproduce**

Load project & slice. Select layer view.

**Actual Results**

All looks good except the layer slider is missing!

**Expected results**

Layer slider should be displayed.

**Additional Information**

[image]

And no, it's not hiding behind the panel!

smartavionics, 2018-10-29:
[cura.log.txt](https://github.com/Ultimaker/Cura/files/2526240/cura.log.txt)

smartavionics, 2018-10-30:
I removed the cache directory and it didn't make any difference.

I notice that it isn't just missing the sliders but also the other controls that affect the layer view.

Vandrasc, 2018-11-01:
FYI @Ghostkeeper, this is what we were discussing yesterday.

Vandrasc, 2018-11-01:
Devs CURA-5885

LipuFei, 2018-11-01:
@smartavionics I made a commit in Uranium and it seems to fix the problem. Could try that?
https://github.com/Ultimaker/Uranium/commit/cbb847a1a2e67858b009abd2db20313e3148c763

smartavionics, 2018-11-01:
Hi @LipuFei , just off to the dentist!, will try later this morning.

smartavionics, 2018-11-01:
Just tried that and it works OK, closing...

closed 2018-11-01 by smartavionics