Ultimaker/Cura #4684: Cura 3.5.1 Fill gaps between walls

Predator75, 2018-10-29:
**Application Version**
3.5.1

**Platform**
Windows 64

**Printer**
custom

**Steps to Reproduce**
Slice the file with 0.4 nozzle and 5 wall count

[RightAnt blu.zip](https://github.com/Ultimaker/Cura/files/2526145/RightAnt.blu.zip)

**Actual Results**
the  gaps between walls are not filled

**Expected results**
Fill gaps between walls

**Additional Information**
[image]

smartavionics, 2018-10-29:
Try turning off the overlap compensation for the inner walls.

Predator75, 2018-10-29:
> 
> 
> Try turning off the overlap compensation for the inner walls.
unfortunately the result does not change

Predator75, 2018-11-02:
No way to fill these gaps? :-(

smartavionics, 2018-11-02:
Have you tried using thinner wall lines?

Predator75, 2018-11-02:
Yes smartavionics, thanks, if I print with a thinner line or I print only 3 walls count the sharp angles are not created.
But why do not it fill them?

Ghostkeeper, 2019-01-22:
In the "Shell" category right under "Fill Gaps Between Walls", make sure that the "Filter Out Tiny Gaps" setting is not checked.

It's checked by default because for many of the smallest gaps, moving the nozzle there and trying to extrude won't have much of an effect since the delay between the extruder and the flow through the nozzle basically prevents any material from being extruded in that tiny amount of time that the nozzle is over the correct spot.

closed 2019-01-22 by Ghostkeeper