Ultimaker/Cura #4686: [3.5.1] New Prime Tower Bug

zerspaner-gerd, 2018-10-29:
**Application Version**
Cura 3.5.1

**Platform**
Windows 10

**Printer**
Tried with UM3 and UM2 dual

**Steps to Reproduce**
Turn on Prime Tower

**Expected results**
At least no gaps.
I'm not sure how it is now.
If both extruders are to print each layer, although one is not needed, this will greatly increase the printing time.

I do not get the same result every time I try. (See pictures)
Sometimes small gaps, sometimes big ones

**Additional Information**
[image]
[image]
[#3DBenchy_-_Dualprint BUG.zip](https://github.com/Ultimaker/Cura/files/2527244/3DBenchy_-_Dualprint.BUG.zip)

zerspaner-gerd, 2018-10-30:
Ok, the difference arises depending on which extruder prints which area.
But even with 5 layer gap it is probably not printable.

[#3DBenchy_-_Dualprint BUG 2ter.zip](https://github.com/Ultimaker/Cura/files/2529458/3DBenchy_-_Dualprint.BUG.2ter.zip)

Ellecross, 2018-11-01:
Hey @zerspaner-gerd I don't think this is bug.

zerspaner-gerd, 2018-11-01:
Hi @Ellecross ,

Nice picture to clarify what I meant, you can not print in the air!

[image]

See red arrows

Either extruder1 must take over, or extruder 2, but this leads to higher print time

zerspaner-gerd, 2018-11-01:
Hi @Vandrasc 

We are talking about the New Prime Tower from Cura Version 3.5

> the prime tower will be then printed only with the extruder in use. Makes sense? I hope it does :)

No it does not, it prints the inner circle in the air

[image]

Ghostkeeper, 2018-11-02:
This is intended behaviour, actually. It might look weird but it's really useful.

The outer shell of the prime tower always prints on every layer. This introduces unnecessary extruder switches sometimes, but is necessary for the tower to keep standing.

The shells on the inside though don't need to print every layer. The idea is that if they print in thin air, the filament is just going to fall down inside the tower. Essentially you've just printed a purge bucket and you're purging inside it. This can save a significant amount of time since you don't need to use every extruder on every layer.

miriad, 2018-12-18:
For a model that has primarily extruder 2 (2 colors, not disolvable support, etc), this new prime tower results in a costly extruder switch (with temperature changes) every layer, just so extruder 1 can print the outer layer of the prime tower.

Can we get an option that the last active (or next active) extruder is the one to print the outer wall? I understand that for disolvable support, it's good to have a nice solid outer layer for the tower, but for dual-color prints, etc, this is just causing unneeded extruder switches, which slow the print down a lot, and also tend to cause under-extrusion when using just extruder 2 (due to the temperature fluctuations)

Hpiz, 2019-01-06:
Don't mean to hijack but this seems a fitting place for this.

Can I make a suggestion that may help the new prime towers (as far as waste)? I get why the inside extruder doesn't prime every layer, that's fine. My issue is with the outside extruder on the prime tower. It prints solid every layer.

Wouldn't it make more sense to have the outside prime tower extruder alternate between solid and support layers? The outside extruder prints it's first prime layer then the first model layer then comes back to do another full prime layer. 

That I suggest should be 10% infill because the extruder is already primed at that point (it's already printed part of the model). This will save 30-40% on that filament for priming as it is currently wasted to support the next layer.

alexjx, 2019-01-30:
Is it possible to make this configurable? I was trying to print a traffic cone with both PLA... however,  the unnecessary nozzle switch does create some artifacts on the layers that do not need a switch.
Nozzle switch has its cost of temperature fluctuation!!!
[image]

Ghostkeeper, 2019-02-01:
> Cura should automatically print the outer circle with the extruder printing the most commonly used. I can not find a way to change that.

It chooses the extruder for the outer shell depending on the material profile. It'll choose the material with the greatest adhesion tendency to print the outer circle width, to maximise the stability of the prime tower. This means that if you have one extruder with PLA and one extruder with TPU, PLA will always get used to print the outer shell. If both extruders have the same adhesion tendency, it'll just choose the first extruder IIRC because the first extruder tends to get used more for the body of the print rather than for support.

> Maybe it would also be possible to make the tower variant variable. Old tower variant before Cura 3.5 / New tower variant from Cura 3.5.

This would require us to maintain two implementations. We won't do that.

> Can we get an option that the last active (or next active) extruder is the one to print the outer wall?

> Wouldn't it make more sense to have the outside prime tower extruder alternate between solid and support layers? The outside extruder prints it's first prime layer then the first model layer (not sure if the second model layer comes before the next prime but) then comes back to do another full prime layer.

The whole point of this new prime tower is to make the tower printed by only one extruder so that there is not any border in the height of the tower. If the tower gets printed by multiple extruders, the tower tends to become very weak because the two materials don't adhere well to each other. I agree that it's a trade-off between reliability and printing speed, but I don't see an easy way to make that configurable that is also clear to users.

AbeFM, 2019-11-05:
I could see a menu item under Dual Extrusion that says:

Prime Tower Backbone (ok, that's a terrible name, sorry):
(Auto)
(Min Material)
ExtA
ExtB
....
ExtN

Auto uses the current, very sane logic.
Min Material would pick a single material but based on material usage. I'm 50-50 on this one, it could be left off.
Ext(X) uses that extruder.

All of them would behave as the current method does - a outer shell of the preferred material - but allow you to pick the one you want. I find the harder TPU's, for instance, to be plenty reliable and would be ok with my tower made of that while others may not.

It seems like a selection like that allows you to over-ride it, sometimes CURA doesn't really know what I'm doing and this would allow me to bypass it while "auto" would work as well as it currently does. It seems straightforward to code though that's an assumption.

Ghostkeeper, 2019-11-07:
One thing we could very easily do is to make the Adhesion Tendency setting visible. It's currently always made invisible (which gives the weird/incorrect message that it's hidden by your printer profile; it's not, but it's hidden by the default settings that your printer profile inherits from). However that makes for a pretty weird setting.

Maybe a setting name like "Prime Tower Outer Shell Extruder" would be better, albeit a bit long. I can see that working.

AbeFM, 2019-11-12:
I think it's a more direct, user-understandable setting. "Adhesion Tendency" looks like a complex, skip-over setting that you tweak without understanding with something unrelated isn't working, you forget about, and effects things down the line.

Prime Tower Main Extruder? Base Extruder?.... Ok, naming things isn't my strong suit.

Ghostkeeper, 2021-06-07:
This had been in our backlog for quite a while waiting to be picked up. However our PO recently rejected it because he says it won't get priority any time soon.

We'll not close this issue since it is a valid issue, but to manage expectations we'll mark this as "deferred" for that reason.

AbeFM, 2021-06-08:
I appreciate the transparency.

AbeFM, 2023-07-01:
Yeah, this one doesn't seem particularly hard (easy to say), but it seems
like there's benefit to doing it. I can't prioritize it, but it's still
valid and useful and should stay on the list in my opinion. Letting some
others go.