Ultimaker/Cura #4687: Cura 3.5.1 slicing glitch

dvanw6, 2018-10-30:
**Application Version**
Cura 3.5.1

**Platform**
Windows 10

**Printer**
Anet A8

**Steps to Reproduce**
Prep the model, slice it, put the SD card into the printer, and print.

**Actual Results**
Printer instantly starts doing what seems to be the model's printing gcode, completely forgetting to do the start gcode which preps everything like heating the nozzle.

**Expected results**
Proper starting of the print, with pre heating and everything.

**Additional Information**
It doesnt happen all the time, and I fixed it by reslicing the model again. By the looks of it, all its saving to my SD card is the model's gcode while forgetting to add the starting gcode for my printer.

nallath, 2018-10-31:
Did you also look at the g-code that it saved to confirm it's not doing this? Could you also share your logs?

khan04what, 2018-11-12:
I have the same problem with sdcard. I save the stl file from cura 3.5.1 to sd card it prints 3 layer, and than the printer makes a blob or the extruder goes back to home and the filament don't stop extrude till end of home axe. I had printfailures 3 times of the same stl. the 4 time print was oke. everytime ill put the sdcard back on pc to cura then to the printer. and my sdcard is an new one and not cheap one, sometimes is this the issue for bad g-code or no code at all.

closed 2020-06-20 by no-response[bot]