Ultimaker/Cura #4693: 3.5.1 zoom stuck @far when inverted in config

Thisismydigitalself, 2018-10-30:
**Application Version**
3.5.1

**Platform**
win10/64, Mouse FlipFlopWheel set to "1" (reversed) in regedit

**Printer**
CustomFDM

**Steps to Reproduce**
In Cura 3.5.1 try zooming **all the way** in/near and out/far. if not stuck, try inverting zoom direction in config and try again. in my case builtplate stuck at far (zoom all the way out) when zoom is inverted in config.

**Actual Results**
zoom stuck at the far end. pressing any of the top camera manipulators releases the zoom. 
[image]
**Expected results**
zoom should never freeze or lock at any stage.

**Additional Information**

nallath, 2018-11-06:
This is a bit to obscure for us to actually try and fix.

closed 2018-11-06 by nallath

Thisismydigitalself, 2018-11-07:
For the record, in my case builtplate stuck at far (zoom all the way out) when zoom is inverted in config.
[ZoomStuck.zip](https://github.com/Ultimaker/Cura/files/2557128/ZoomStuck.zip)

Thisismydigitalself, 2018-11-13:
Same with 3.6 released version. zoom stuck either zoomed all the way in and/or out.