Ultimaker/Cura #4694: have to power cycle printer after an autoleveling failure

sisu-jason, 2018-10-30:
**Application Version**
4.3.3.20180529

**Platform**
ultimaker 3 extended

**Printer**
ultimaker 3 extended

**Steps to Reproduce**
This most often happens when the breakaway support material is loaded in one of the printcores. Breakaway support material often leaves a bit of material extruded out of (and stuck to) the nozzle. When the auto-bed leveling process for the next print is initiated it will fail because the difference between printcore heights is out of spec (because of the extra material stuck to the nozzle). When this happens the printer fails with a message to go to ultimaker.com for support. Really, all that needs to be done is to clear the extra material stuck to the nozzle and then restart the print. The problem is that 

**Actual Results**
After I clean the nozzle of excess material stuck to it I have to power-cycle the printer before I can start a new print.

**Expected results**
I would like to be able to clear the error message and retry the auto-leveling (after cleaning the excess material off of the nozzle.) 

**Additional Information**

nallath, 2018-11-06:
This is only for Cura bugs and/or feature requests.

closed 2018-11-06 by nallath

sisu-jason, 2018-11-06:
That's fine. Could you please direct me to where I can log Ultimaker
firmware issues?

Thanks,

Jason

nallath, 2018-11-06:
I will forward our devs here since I'm not aware of any official place to do it.