Ultimaker/Cura #4696: [Feature] Do not remove layer view when placing support blockers

MB3Y, 2018-10-30:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

alekseisasin, 2018-11-13:
Hi @MB3Y , Can you make some drafts to understand the "Actual" and "Expected" results? It will help to the proposed improvement.

MB3Y, 2018-11-19:
Hi @alekseisasin , thanks for  @asking. Find below additional "Actual" info an pictures. I hope that from this also the "Expected" is more clear. Normally I would say that I cannot include "Expected" screenshots, but there is this small bug since the addition of the "Play" feature that shows "gray layer information" ;).

Slicing [this particular model](https://www.thingiverse.com/thing:1099700) takes 15s for me. I don't need all that support everywhere and even when reducing the angle some support is still there on small bridges or overhangs that seem redundant for me. So I started putting support blockers. These had to be precisely located to keep one pillar of support, removing another adjacent one.
To know where the unwanted support is placed -and hence to know where to put the support blocker- I need to be in the "Layer view".
When I need to put 4 blockers, I need to reslice each time I place the blocker as well as several times resizing the blocker to it's right size.

**Sliced first time**
[image]

**Placing first blocker on left horn of the mask** --> layer information gone (is the blocker in correct position? --> need to reslice)
[image]

**Resliced** - Oh... blocker is not OK yet --> resizes/reposition/rotate... each time an action is done the layer view is gone and one can never know for sure the support blocker is covering the support since the support is not visible anymore when manipulating the support blocker block
[image]

**While resizing the support blocker** - layer view gone
[image]

In this particular model I've placed half a dozen of support blockers making a lot of iterations like this...

"Expected" suggestion: do keep the indication that reslicing is needed, but don't indicate this by removing the "Layer view" information completely, show the invalid "Layer view" grayed out like e.g. in the next picture
[image]

alekseisasin, 2018-11-20:
We discussed the PR and it has a reason to be implemented. The only thing that it will take some time before Cura will start work on it.
DEV: CURA-5958

MB3Y, 2018-11-20:
@alekseisasin Don't worry about that - filling the product backlog is also needed. Patience is a virtue ;)

Ghostkeeper, 2020-06-30:
This one is still in our backlog. It's been estimated so it must have been on the short-term planning, but sunk back down afterwards.

Schokobecher, 2020-08-28:
Very nice, looking forward to this. Maybe in 4.8?