Ultimaker/Cura #4697: A way to zoom in on small or flat objects

jaxelsson, 2018-10-30:
Currently I can't seem to zoom in on the print bed itself, to view smaller or flat objects.  This is a big hindrance.  Would be good if the center point of the zoom was by default the center of the object or the bed, instead of 50% of the way up into thin air.  (See screenshot)  

Love the app, but this is becoming a deal breaker for printing small/flat things, as I sort of just have to hope for the best without inspecting the path.

[image]

Thisismydigitalself, 2018-10-31:
Have you tried this?
[image]

alekseisasin, 2018-11-13:
@jaxelsson , Did you try the setting "Center camera when item is selected"

GregValiant, 2024-11-24:
@jaxelsson is this still relevant in new cura versions (5.8.0 and up).
I'll close this.  Let me know here if you feel that is premature.

closed 2024-11-24 by GregValiant