Ultimaker/Cura #4698: 3.5.1 crashes while activating a printer (crash log attached)

Thisismydigitalself, 2018-10-31:
**Application Version**
3.5.1

**Platform**
win10/64

**Printer**
Custom FDM

**Steps to Reproduce**
Adtivate TLM_#2 printer (probably an instance from previously loaded project)

**Actual Results**
Cura crashes. log file attached
[cura.log](https://github.com/Ultimaker/Cura/files/2533053/cura.log)

**Expected results**
Crash is not an expected result.

Eyos, 2018-10-31:
I have the same problem. Downgraded back to version 3.5 because of it.

Thisismydigitalself, 2018-10-31:
Why Activating a specific printer instance is loading a specific (linked?) profile? what's the reasoning behind this?

Eyos, 2018-10-31:
The same here. I have few printers with many different profiles for each.

Vandrasc, 2018-11-01:
Hi all, this happens each time you are activating a printer? Would it be possible to share your cura user folder with us? Unfortunately, we are not able to reproduce with just the above steps and we hope that by using the same configurations as you, we will be able to find the culprit here and help you as much as we can. 
 
Thank you!

diegopradogesto, 2018-11-01:
I see this crash in the logs:
```
2018-10-31 08:38:09,860 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-10-31 08:38:09,862 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-10-31 08:38:09,864 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 70, in getOverridesForExtruder
2018-10-31 08:38:09,865 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 193, in _settingIsOverwritingInheritance
2018-10-31 08:38:09,867 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'getAllKeys'
```

And this was indeed fixed by @nallath when doing some boy scouting in https://github.com/Ultimaker/Cura/commit/6eb8b754903f3fe1db5baa55ad90305fda94de38#diff-b81a580f9fe02a76a0a39c678effe256R175

It shouldn't happen anymore now. Thanks for reporting.

closed 2018-11-01 by diegopradogesto