Ultimaker/Cura #4702: Cura crashes on saving files

coldgas, 2018-10-31:
Application Version
3.5.1 & 3.6.0 (Beta)

Platform
Windows 10 Home

Printer
Creality Ender 3

Steps to Reproduce
Hit save to file

Actual Results
Crashes every time without saving a file. (3.5.1)
In 3.6.0 it starts writing the file and stops after 32kb written and crashes.

Expected results
Save G code to file

Additional Information
Workaround: Disable "(Send anonymous) print information" on general settings fixes the problem in both versions.

LipuFei, 2018-10-31:
@coldgas I discussed with @diegopradogesto and we think this could be related to issue https://github.com/Ultimaker/Cura/issues/4586. It can have something to do with Windows Defender. Could you try adding Cura.exe as an exclusion in Windows Defender like suggested in this comment https://github.com/Ultimaker/Cura/issues/4586#issuecomment-432061598?

LipuFei, 2018-11-06:
Due to the lack of response, I will close this issue.

closed 2018-11-06 by LipuFei