Ultimaker/Cura #4707: Slicing issue (?) printer moves to corner and extrudes filament

osterac, 2018-11-01:
Sometimes when I try to print with my Creality Ender 4 my printer will, after heating and homing, move to the opposite corner (while slightly adjusting the z axis) and extrude massive amounts of filament until I cancel the print (SD card) or unplug it (printing from USB - abort print doesn't work). if I'm printing from USB, Cura will then crash.
[v beam x2.zip](https://github.com/Ultimaker/Cura/files/2539261/v.beam.x2.zip)

**Application Version**

3.5.1
**Platform**

Windows 10 x64
**Printer**

Creality Ender 4
**Steps to Reproduce**

Seemingly random occurrence upon printing a project
**Actual Results**

 my printer will, after heating and homing, move to the opposite corner (while slightly adjusting the z axis) and extrude massive amounts of filament until I cancel the print (SD card) or unplug it (printing from USB - abort print doesn't work). if I'm printing from USB, Cura will then crash.
**Expected results**

Normal print of my project
**Additional Information**

smartavionics, 2018-11-01:
I sliced your project and looked at the gcode (using craftware) and I can't see any obvious reasons for the behaviour you describe. Here's the first layer:

[image]

osterac, 2018-11-02:
Well I hope it's not my printer... If it is, it's something that's only started recently.

osterac, 2018-11-02:
It's happened with other projects too

Ghostkeeper, 2019-02-05:
I'm also unable to reproduce this, even with your project file. Maybe this was fixed in a more recent version?

osterac, 2019-02-06:
It may have been something with the firmware. I've since stopped using the stock firmware and switched to straight Marlin with my printer's parameters added. I've had no problems like this since.

closed 2019-02-06 by osterac