Ultimaker/Cura #4708: 3.6B crashes when loading 3.5.1 projects

Thisismydigitalself, 2018-11-01:
**Application Version**
3.6B

**Platform**
Win10/64

**Printer**
CFFFP

**Steps to Reproduce**
Load 3.5.1 *3mf project.

**Actual Results**
3,6B crashes.

**Expected results**
3.6B should load 3.5.x projects. this is basic. for me, not being able to open 3.5.x projects renders 3.6B useless.

**Additional Information**
[cura36b.log](https://github.com/Ultimaker/Cura/files/2539317/cura36b.log)

Thisismydigitalself, 2018-11-01:
Update:
When i chose to create new printer and new profile - 3.6B does not crash so now i need to check what got changed in my printer profile.

I see cooling fan was added, and as you can see i have new printer instances _#2 _#3 _#4 .. and for _#2 nozzle size is 0.0. 

I must say that there was not even one version of Cura these last 24 months or so that didn't have issues with nozzle size. always issues with nozzle sizes. This really is frustrating and disappointing. 

[image]

Thisismydigitalself, 2018-11-02:
Now i see all my 3.5.1 imported profiles to 3.6B are locked and i can't slice stl's.
something went very wrong during importing projects. my guess is: something is broken between printers and profilles. now none of my 3.6B printers/profiles are usable. 3.4.1 was good to me. 3.5.1 is what i am using these days. as much as i like 3.6 new features i will have to stick with 3.5.1 for now. 

3.6B wont slice anymore.
[image]

LipuFei, 2018-11-06:
Hi @Thisismydigitalself , could you upload the project file for this issue?

LipuFei, 2018-11-06:
Fixed in #4742

closed 2018-11-06 by LipuFei