Ultimaker/Cura #4710: Suggestion: Seperate imported Projects / Printers from those created by user.

Thisismydigitalself, 2018-11-01:
**Application Version**
Any

**Platform**
WIn10/64

**Printer**
CFFFP

**Steps to Reproduce**
import *3mf projects

**Actual Results**
messy profiles and printers dbase

**Expected results**
New tab / separators for seperating my clean dbase list of profiles and printers from those creates when openning other any other projects.
 
Cura list of printers and projects is getting very messy.

diegopradogesto, 2018-11-08:
I don't really understand what do you mean. Are you talking about the dialog that is opened when opening a project file?

In the preference panel, you have a list of printers, materials, profiles, ... but you probably know about it, that's why I don't know what you are suggesting. Could you elaborate your idea a bit more?

Thisismydigitalself, 2018-11-11:
Hi there,
I wanted to screen capture my Cura but any screenshot sw i use closes Cura's open menus during snapshot so I can't show you but i can try to describe the challange:
I got two CFFFP printers, hence only two printers configured but when I import legacy project or open other projects, Cura creates new printers instances, some with #2_ #3_ #4_  etc suffixes and maybe other printers used during project creation which all of these are now part of my Printers & Profile's one long list and this creates a mess which i would like to eliminate.  MY wish is for every project i import to be able to chose a global folder/enviorment/workspace/etc.. so i can filter those out by tickbox/tab to see whatever i wants to see.  i hope this wish of my makes sense.

diegopradogesto, 2018-11-11:
When the dialog opens, you can decide whether you want your printers and profiles to be overwritten or create a new one. Of course if you select to create a new one, Cura will create a new one by adding the suffix. If you do that many times, you will get a long list of printers or profiles. For this and many other reasons, the Printer and Profile manager exists. You can delete them in the same window.

It's my opinion but I don't see a problem here, it depends on how you manage your printers and profiles. Also I don't see any benefit on that. I would suggest to clean your list of printers every now and then if you use the option "Create new" when importing project files.

Thisismydigitalself, 2018-11-11:
I don't see a problem either but only a good way to make some order in live database that keeps changing. I am pretty sure it is not that hard to add another checkbox in this project importer to flag projects it in such a way they will be hidden and/or in slanted font. From past experience i am 100% sure this request will soon be closed following this statment:  "Our project manager removed this from our planning because it was not seen as important enough."

True, but nice to have.

alekseisasin, 2018-11-13:
If I understand correctly you want to define workspace for every loaded project. During loading, you want to control in which workspace the project will be saved. Then a new feature would be: "Allow to change working workspace".

Thisismydigitalself, 2018-11-14:
Yes, more like a default one and an option to chose if to open the project in other workspace so it wont be visible in my own dbase.

**Here's my Printer's list. only two are my own. others were created during import.**

[image]

**Here's my Profile's list. My profiles are in [ ], the rest imported from others.**

[image]

Just a suggestion for keeping order in our DB's.
Thx.

diegopradogesto, 2018-11-20:
Discussed with the rest of the team and we decided not to do this. However I'll mark it as a possible new feature.

As an idea, there could be an option in the workspace load dialog asking whether you want to persist the created container. The only drawback is how to deal in case you do some changes in the container and then in the next startup you could get "Configuration errors"

GregValiant, 2024-11-24:
I'm cleaning house around here and this has been open for several years.  I'll go ahead and close this.  I hope you understand.

closed 2024-11-24 by GregValiant