Ultimaker/Cura #4711: remains of a previous slice on a new slice

brunatort1101, 2018-11-01:
**Application Version**
3.51

**Platform**
windows 7 64bit

**Printer**
anet A8 

**Steps to Reproduce**
First I load the 30x15x14.stl file. make 14 of them on the same slice and save them on the SD, after that I delete all of them and load the panel vuelo.stl file, slice and save. go to print the second one and gets destroyed by the nozzle sinking on the print. checked out the layer wiew with nozzle movement and see the first file in the middle of the second 

**Actual Results**

[image]
[image]

[30x15x14.zip](https://github.com/Ultimaker/Cura/files/2540393/30x15x14.zip)
[panel vuelo.zip](https://github.com/Ultimaker/Cura/files/2540396/panel.vuelo.zip)

Vandrasc, 2018-11-02:
Hi @brunatort1101, could you please share the faulty gcode with us? Unfortunately, we are not able to reproduce the issue as described by you, I tried it on a CFFF printer. Is this happening each time or one time only? Many thanks!

brunatort1101, 2018-11-02:
[panelbotones.zip](https://github.com/Ultimaker/Cura/files/2544126/panelbotones.zip)
thsi is the gcode of that failure

alekseisasin, 2018-11-13:
It is strange issue, we cannot do much with it. Because we do not have the  anet A8 printer, 
In attached gcode indeed I see the same result as on your image. 
I tried to do same steps from "Steps to Reproduce" using Ultimaker 3 printer and I god correct Gcode. You can make a video from your computer screen, maybe you did some extra steps which I miss. Then maybe we can understand the issue better. 
I have used Cura 3.6 version.

closed 2020-06-20 by no-response[bot]