Ultimaker/Cura #4712: [3.5.1] Slicing error: Random movement and extrusion to corner mid print

mscosti, 2018-11-02:
**Application Version**
3.5.1

**Platform**
Mac OSX

**Printer**
Creality Ender 3

**Steps to Reproduce**
Sliced an STL file, gave it a name, and saved it straight to my removable drive (which contains the micro SD).

**Actual Results**
Print seemed to be doing great, but then after getting through a fair amount of layers (10-15 or so) the printer moved to the lower left hand side of the bed while actively extruding filament, and made its way back and continued as if nothing had happened. After the print, I analyzed the .gCode I used closer and viewed it at http://gcode.ws/ and noticed the move while stepping through layer by layer. I also opened up the .gcode myself and interestingly, there is *no* layer 13 defined in the .gCode at all, around where the incident happened. 

**Expected results**
Prints all expected layers, without random moves across the bed, and potentially skipping a layer.

**Additional Information**
.gcode sliced with cura 3.5.1 and saved onto memory card: https://drive.google.com/file/d/1NHSNI9dCJey0ZJN9hzlsU4zaEFN1D82h/view?usp=sharing

Interestingly, The .gCode is not able to be opened in cura, it says it is failed to load to due to being an invalid file.

I'm still new to this, and didn't realize the utility of saving entire cura project files, and I apologize for not having one. Going forward I'll save my projects, and can include if I encounter this again.

Attached is a screenshot from the .gCode simulated in gcode.ws.
[image]

Attached is a photo of what the move looked like in the actual print.
[image]

mscosti, 2018-11-02:
Update:
I just sliced another model and opened it up in the same gCode viewer and noticed what may be a similar issue. I have saved the project file this time

screenshot: 
[image]

[fidgetCube.3mf](https://drive.google.com/file/d/1N4Mp1wRyFmjRxM5AEW7o1aVT3HTKX8pl/view?usp=sharing)

[fidgetCube.gcode](https://drive.google.com/file/d/18iQyYu6F4zCC6jOTtTXak1aPzs7uLJtT/view?usp=sharing)

smartavionics, 2018-11-02:
Your first file is corrupt and doesn't load correctly into 2 of the 3 gcode readers I tried.

I could not see any problems with your second gcode file.

Ghostkeeper, 2019-02-05:
Cura gives this error when loading the first g-code file:

    UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 720896: invalid continuation byte

Something fiddled with that file.

The second file (the fidget cube) is expected behaviour though. It makes that travel move because the end g-code contains this line:

    G1 X0 Y{machine_depth} F1000 ; Move Heat Bed to the front for easy print removal

So the idea was that this makes it easier to remove the print.

This is different from your first bug though, but just like SmartAvionics in November, I can't reproduce this problem.

closed 2020-06-20 by no-response[bot]