Ultimaker/Cura #4713: Hollow bell is made solid

iainmack52, 2018-11-02:
**Application Version**
3.5.1

**Platform**
Windows 10 Home 64 Bit

**Printer**
As soon as I figure out how to attach - I will. Printer is Anet E10

**Steps to Reproduce**
Just import STL and slice. I tried several different SketchUp and Blender files, creating different STL files. They all had the same result.

**Actual Results**
A hollow bell becomes an object that prints with a solid/covers the whole base layer.

**Expected results**
I expected to see a circle of print, becoming a bell shaped object. This used to happen with this printer on this PC a month or so ago. May have been before 3.4 even?
[AnetSketchUpBellProject.zip](https://github.com/Ultimaker/Cura/files/2541242/AnetSketchUpBellProject.zip)

**Additional Information**
On the screen it all looks just fine, including in Cura, but this file is coming out with a complete layer rather than just the edges, meaning the body of the bell. The only thing that has changed is Cura. Blender is the same version, SketchUp is the same version, the Anet E10 has not been changed or updated in any way. I don't want a bottom layer in the central void of the bell.

smartavionics, 2018-11-02:
Turn off the Union Overlapping Volumes setting...

[image]

smartavionics, 2018-11-02:
This is what the mesh tools plugin thinks of your model...

[image]

iainmack52, 2018-11-02:
Thank you Mark. You can probably tell I am new at this - and didn't even
know that option existed.

Regards,
Iain

smartavionics, 2018-11-02:
Yes, but if your model was good you would not need to change that option. The model needs fixing.

iainmack52, 2018-11-02:
No idea. None of the models used were mine.

rburema, 2018-11-07:
Since this is an issue with the model, rather than with Cura, I'm closing the issue.

closed 2018-11-07 by rburema