Ultimaker/Cura #4714: Repetier Crashes because of Cura interference!

gnuion, 2018-11-02:
The issue is simple and I believe it will happen to all printers. When using Repetier to run the gcode and the printer is USB connected, Cura tries to take control of the USB connection therefor crashes Repetier destroying the print. it really makes me angry when I need to open Cura, forget about Repetier and my print stops!

smartavionics, 2018-11-02:
Hello @gnuion , I'm sorry I can't help directly with your issue but the people who could help you would really like to know some more info like what OS you are using, Cura version, etc. Thanks.

fieldOfView, 2018-11-02:
You can disable the USB Printing plugin in Toolbox/Marketplace on the "installed" tab. This will disable any USB trickery by Cura.

gnuion, 2018-11-02:
fieldOfView thank you! I can mark this as solved now!

fieldOfView, 2018-11-02:
Cura shouldn't crash other applications even with USB Printing enabled. Having said that, if repetier is crashing because it is letting Cura take over the serial port, then that would be strange behavior from repetier. Like @smartavionics says, what OS and version(s) are you using?

gnuion, 2018-11-02:
I am using Windows 10, Repetier V2.1.3, Cura 3.5.0. Had the same issues on older Windows OS, Cura and Repetier versions. Didn't try it under Linux.

gnuion, 2018-11-02:
It doesn't happen right away though, usually the print stops after 30 minutes. If I don't open Cura while Repetier is running the print finishes even if it takes a day. So it took time for me to notice the cause.

closed 2018-11-03 by gnuion