Ultimaker/Cura #4716: Cura 3.5.1 do not respect standby temperature when changing from extruder 1

dtuzhilin, 2018-11-02:
when start pronting with extruder 2 gcode does change extruder 1 temperature to "Final Printing Temperature" rather than to "Standby temperature" as defined per material definition. However, estruder 2 standby temperature is considered and extruder 2 material does not get overheated.

diegopradogesto, 2018-11-02:
Next time, please fill out the form. We need some extra information in order to help you. Please provide Cura version that you are using, clear steps how to reproduce the issue and a project file.

dtuzhilin, 2018-11-02:
Sorry, Diego. All is written. Cura 3.5.1, two extruders, when printer switch to 2nd extruder, temperature of first extruder falls to value of "Final Printing Temperature" (225 C in my case) instead of going to "Standby temperature" (150C as defined in material profile). This is wrong - leads to material leaking from extruder.
When printing switch back to 1st extruder - temperature of 2nd goes to it's "Standby temperature". That is correct behaviour.

diegopradogesto, 2018-11-02:
Please send project file.

dtuzhilin, 2018-11-02:
[CFFFP_minnie.zip](https://github.com/Ultimaker/Cura/files/2544226/CFFFP_minnie.zip)

LipuFei, 2018-11-06:
Hi @dtuzhilin I loaded your project and sliced with 3.6 BETA, the problem seems to be fixed. I think this could be related to #4567. Could you try 3.6 BETA see if this is still a problem? Thanks.

BBuenzow, 2018-11-08:
Hi.  New Cura user here.  I recently updated to a dual extruder RepRap-type FDM that came with Cura 3.3.1.  Have updated to 3.5 but standby temperature issue remains.  We came up with our own "Rosetta Stone" to get the correct temperature inputs to the FDM by a lot of trial and error:

The standby temperature of extruder 1 comes from the printing temperature of extruder 2.
The printing temperature of extruder 2 comes from the default printing temperature of extruder 2.

Hopefully this will help out others while the Cura program gets debugged.

dtuzhilin, 2018-11-10:
I did not noticed improvement. Even worse - 3.6 is slicing infill and walls like ebfore (makng infill after walls) - this makes infill strictures visible on 2 layer default walls. Previous version (3.5.1) makes is better: walls after infill.

Ghostkeeper, 2018-12-19:
Currently (Cura 4.0) the provided project file results in an assertion failure in the engine. We've previously seen that this assertion failure, when ignored in release builds, results in faulty travel moves.

I'm suspecting that this might be a case of the Minimal Time Standby Temperature. If the extruder is used for less than a certain short time (by default 50 seconds) we don't go to the standby temperature at all because PID regulators for the temperature control tend do go whacky when you switch temperatures big and often.

closed 2020-06-20 by no-response[bot]