Ultimaker/Cura #4718: Gyroid Is great! one small thing thou..

Thisismydigitalself, 2018-11-02:
**Application Version**
3.6B

**Platform**
Win10/64

**Printer**
CFFFP

**Steps to Reproduce**
Use Gyroid infill - it's great!

**Actual Results**
Not sure if this is an issue, but infill overlap % does not effect Gyroid infill. it does influance other infill patterns.

**Expected results**
To be able to control overlap. 

**Additional Information**
Thank you for Gyroid Infill @smartavionics and all.

smartavionics, 2018-11-02:
Ah, yes. I forgot about that. Sorry, will take a look over the weekend.

smartavionics, 2018-11-03:
Fixed that, thanks for the report.

Thisismydigitalself, 2018-11-03:
[image]

smartavionics, 2018-11-03:
Nice colour! Is that PETG? I use it a lot and find I get very good results. I use no-skin combing with retracts for travels > 10mm.

Thisismydigitalself, 2018-11-03:
You are an observant person. 
Combing was Off. i've now updated my profile to no-skin. 
This is red PLA+. i find PLA+ colours to be more vivid than regular/cheap PLA's. overall much better stuff.

smartavionics, 2018-11-03:
I would also set the max combing distance without retraction to something like 10mm as that allows short non-retract travels but retracts for the longer moves which reduces the dribble.

Thisismydigitalself, 2018-11-03:
Retraction, on my direct drive system, was set to 5mm . 10 or 5 does not change print time so if 10 wont worsen things up while cutting retractions in half, than 10 would be the new 5.

smartavionics, 2018-11-03:
I don't mean the retraction distance, I mean the Max Comb Distance With No Retract setting.

Thisismydigitalself, 2018-11-03:
I did not even know this setting exists and so never used it before.
Wanted to try it out on this same project but guess what - 3.6B crashes when i try to load it back.
No joy.

smartavionics, 2018-11-03:
Ha! that's Cura for you, each release is guaranteed to provide both good and bad changes!

Thisismydigitalself, 2018-11-04:
One cannot appreciate good with and/or without bad.
Curadox.

fieldOfView, 2018-11-04:
Totally off-topic, but WAT?
[image]

Thisismydigitalself, 2018-11-04:
> 
> 
> Totally off-topic, but WAT?
> [image]

A glitch in the Matrix

smartavionics, 2018-11-04:
Hmm, does that mean that we can now submit PRs for bugs that haven't occurred yet?

Thisismydigitalself, 2018-11-16:
@smartavionics 
Question: mm/s wise - when infill wiggles like with Gyroid, what is being considered as the legth of it?
the overall length of the line or th top footprint end to end? Sometimes i feel the need to reduce infill speed cause in tight corners the printhead breaks my printer's speed limit.

smartavionics, 2018-11-16:
Hi @Thisismydigitalself , the length is going to be the sum of all of the individual line segments that make up the gyroid curve. All of the line segments are printed at the same speed (the infill mm/s). I don't know why a tight corner should be a problem. What I have seen is that resonance can occur when the frequency of the gyroid wave matches the printer's resonant frequency. In that situation the print head can thrash around rather alarmingly. The solution is to either reduce the mm/s or density as that lowers the gyroid wave frequency.

diegopradogesto, 2019-03-08:
The fix for this bug was added to master.

closed 2019-03-08 by diegopradogesto