Ultimaker/Cura #4719: cura Schmiert mir immer ab beim profil wechsel

Razaek, 2018-11-02:
Also so Langsam nervt es mit Cura.Ich hatte Drucker Probleme weil ich zu viel geupdatet hatte und die alten Versionen nicht entfernt habe.
Folgendes Problem:
Der Drucker lies sich mit der Software nicht mehr richtig configurieren. konnte werte einstellen nur nach 2 drittel der Druckzeit rauscht mir die Nozzle voll ins Druckteil. Die Software schmiert ab wenn ich zb von pla auf petg Eintellung wechsel.Cura schliesst sich einfach.
Jetzt habe ich alles gelöscht inkl. Ordner und habe mir meine Einstellungen notiert. Cura neu auf gespielt und es lief 1 a. keine Probleme mehr.
Nach 4 Modellen wollte ich heute ein neues Modell slicen mit cura 3.5.0 und zack wieder derselbe mist.Cura schliesst sich beim Profilwechsel.ich habe nichts geupdatet nichts verändert.

Also was soll ich nach eurer Meinung machen?

rburema, 2018-11-07:
Hi @Razaek,
You could try to see if the issue is resolved in 3.5.1 or 3.6.0-BETA. We fixed a few crashes between then and now. If this doesn't help you should post the log file here, so we can get a clue as to what might have gone wrong.
cheers, Remco.

diegopradogesto, 2018-11-08:
And... BTW, next time it would be great if you can post the issue in English. That would help many other users with the same problem if we find the solution.

closed 2019-03-01 by nallath