Ultimaker/Cura #4720: 3.6beta printhead shadow too big

PeggyB3D, 2018-11-03:
**Application Version**
3.6 beta

**Platform**
 
mac10.13.6

**Printer**
 
S5

**Steps to Reproduce**
 add a small object and use 'one at a time'

**Actual Results**
 shadow of the printhead is huge, multiplying objects is almost impossible

**Expected results**
size should be as in previous versions

**Additional Information**

[image]
[image]

nallath, 2018-11-06:
We already had this logged on our internal tracker (CURA-5822)

LipuFei, 2018-11-19:
Fixed in https://github.com/Ultimaker/Cura/commit/93bd5fee5387e45ea880ee1fc6125790546e465a

closed 2018-11-19 by nallath

JeremyDWilliams, 2019-04-16:
Still seeing this issue with 4.0.0

Ghostkeeper, 2019-04-17:
We'll need some new reproduction steps if you're still seeing it, because I can't reproduce the issue in 4.0 either.