Ultimaker/Cura #4721: [3.2.21] Angled infill from side walls

kenatmit, 2018-11-03:
I had an idea for a new type of infill. For objects like a tall box, the infill could be built out from the side walls at a 45 degree angle near the top of the inside wall. This prevents needing any infill in the bottom part of the object.

For example, to build a hollow cube 8cm tall x 4cm x 4cm. The supports from the sides need to come in 2cm to reach the middle of the top. The Infill would start at a height of 6cm and angle in at 45 degrees.  The volume above the 45 degree plane could be filled with any other infill method.

The angle can be chosen based on the printers ability to print overhangs.

This type of infill could be thought of as concentric 3D starting at 6cm at the inside surface of the object and then getting smaller at each layer. 

**Application Version**
3.2.21

**Platform**
Windows 7

**Printer**
Lulzbot Taz6

fieldOfView, 2018-11-03:
3.2.21 is a Lulzbot Edition of Cura, which has significant changes from Ultimaker Cura. Please refer to them for support of their version of Cura.

diegopradogesto, 2018-11-08:
I'll close the issue since it is not related with our own version of Cura.

Ghostkeeper, 2019-02-06:
The issue is just as valid for our version of Cura though.

kenatmit, 2019-02-08:
Correct. This is a general idea for improved infill and is not specific to Lulzbot.

Ghostkeeper, 2019-02-12:
Did you try combining "Infill Support" with "Tree Support"? It does something like this.

closed 2026-09-06 by GregValiant