Ultimaker/Cura #4723: [3.5] Hair tip starts in hair and there isn't a support

Liger0, 2018-11-03:
**Application Version**
3.5.0

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use supports (normal or tree) on a model with little hair tips starting in hair.

**Actual Results**
Some of the parts starting in hair won't be supported, the others will have support too much distant that will be useless

**Expected results**
There should be a support

**Additional Information**
[image]
[image]

diegopradogesto, 2018-11-08:
Can you please provide the STL or (even better) the project file?

Liger0, 2018-11-08:
> 
> 
> Can you please provide the STL or (even better) the project file?

[CCR10S_tree support.zip](https://github.com/Ultimaker/Cura/files/2563522/CCR10S_tree.support.zip)

diegopradogesto, 2018-11-10:
I was taking a look, it seems that this tiny part is skipped by the algorithm because is very small and very close to the main body. Using normal supports with default parameters would also skip this part for the same reason. You can maybe try reducing the X/Y distance, increasing the support horizontal expansion, ... But I don't know if it will be an easy fix for that. We appreciate you post the issue, since the Tree Support is experimental we want to keep improving it.

BTW, I checked the model and I saw some weird parts, such as the hair is a different part than the rest of the body and some other surfaces crossing each other. See pictures:

Artifacts in the model due to connected submeshes:
[image]

Information from the Mesh Tools plugin:
[image]

Where the two parts cross each other:
[image]

alekseisasin, 2018-11-12:
Hi @Liger0 , 
To solve this case you can try to change the setting: Support X/Y Distance. From 0.7 to 0.1, Then extra support will be generated. As well the change increases the printing time and material usage as well.
The second option you can try to add a small block with and set Per Model Setting "Print as support" and place next to the place where you need extra support. 
I hope it will help you to improve the quality of the printed model.

Liger0, 2018-11-28:
Anothere edge-case similar to the already mentioned one but regarding the Z distance instead of the X/Y is  the behaviour you can see when the Z distance is very low (1 layer as an example), in which case using the tree supports won't touch it.
I guess it is so because the option to make the supports also on the model and not only on the build plate doesn't apply to the tree supports so the needed supports can't go over the hidden parts. 

If it is how I think, I see the only solution would be an intelligent support function which would detect any unsopported point and tell you the tree support is not suited for the model.

Here I'll attach an example.
[image]

[support tree layer.zip](https://github.com/Ultimaker/Cura/files/2626341/support.tree.layer.zip)

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]