Ultimaker/Cura #4724: Extruder Start\Stop Code, Extruder Switch Retraction

dylanetaft, 2018-11-03:
Cura 3.5.1

Windows 10

Custom Printer

**Steps to Reproduce**
Two things

1) Extruder Start\End GCode 
It is not documented here when it is executed
https://ultimaker.com/en/resources/52029-manage-printers
For example, set some custom gcode on extruder start
G92 E0
G1 F800 E80 ;advance filament 80mm

Expected result:
The gcode put in extruder start would be executed right after printer start gcode.  

Actual Result:
The code appears to only be executed on extruder switch - meaning extruder 0 start isn't ever called on pritner startup, extruder 0 end code called on extruder switch.  

2) Nozzle switch retraction distance 
Set it to 70mm

Expected result: A specific order of operations in the gcode during T1 T0 gcode operations
Extruder 0 start: No advancement from this setting
Extruder 0 end: Retraction of 70mm
Extruder 1 start: Advancement of 70mm
Extruder 1 end: Retraction of 70mm
Extruder 0 start: Advancement of 70mm

Actual results:
Extruder 0 start: No advancement from this setting
Extruder 0 end: Retraction of 70mm
Extruder 1 start: No advancement from this setting **
Extruder 1 end: Retraction of 70mm
Extruder 0 start: Advancement of 70mm
The setting only has an effect on the second nozzle switchover

Other information:  This could be intentional, to support normal two extruder printers.  However, it is not usable for single hot end, single nozzle Y splitter bowden printers.  

Workaround:
Don't use nozzle switch retraction distance setting on bowden single nozzle Y splitter printers.  Instead, load the printer with both filaments the same distance out of the hotend - in the case of my printer, Creality CR-X, 70mm.  In the printer start code, select extruder 0 and advance the filament 70mm.  In both extruder start\end code, advance and retract the filament 70mm.  This will result in expected behavior - that each extruder when used will advance 70mm, and then retract 70mm on switch.  The wipe\purge\prime tower works to prime each nozzle on switchover with no changes.

dylanetaft, 2018-11-04:
Would it be appropriate to put this behavior in the tooltip?
Ie, in the extruder gcode, if you hover over the text box, it says "Start g-code to execute whenever turning the extruder on"

If this behavior is expected, it should be "Start g-code to execute whenever this extruder is switched to"

And Nozzle Switch Retraction Distance currently reads 
"The amount of retraction: Set at 0 for no retraction at all.  This should generall be the same length of the heat zone"

Maybe "The amount of retraction: Set at 0 for no retraction at all.  This should generally be the same length of the heat zone.  Cura will generate gcode to both retract and extrude filament by this amount, but only on the second nozzle switch.  This is typically used for dual nozzle printers, not bowden single nozzle"

Someone can wordsmith that better.

nallath, 2018-11-16:
I've updated them here; 240ac1f06b6458d154d84ce326a3dd8e16777aa1