Ultimaker/Cura #4725: [3.5] Empty part

Liger0, 2018-11-03:
**Application Version**
3.5.0

**Platform**
Win 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Load the file and try to slice

**Actual Results**
The part on the head is empty.

**Expected results**
The part on the head is full.

**Additional Information**
[image]
[image]
[CCR10S_BL.zip](https://github.com/Ultimaker/Cura/files/2545417/CCR10S_BL.zip)

smartavionics, 2018-11-04:
If you use a wall line width of 0.3 or less then those head walls are printed, they must be very thin.

[image]

Liger0, 2018-11-04:
> 
> 
> If you use a wall line width of 0.3 or less then those head walls are printed, they must be very thin.
> 
> [image]

Hi there. Why doesn't it fill in instead?

smartavionics, 2018-11-04:
Because the model isn't solid?

Liger0, 2018-11-05:
I found the errors in my model, it seems there were 2 vertices not needed. Thanks.

closed 2018-11-05 by Liger0