Ultimaker/Cura #4726: Cura 3.6.0 beta stops printing too soon and under extrusion last internal wall

Predator75, 2018-11-04:
**Application Version**
3.6.0 beta

**Platform**
win x64

**Printer**
cfff

**Steps to Reproduce**
1.75 filament diameter, 0.4 nozzle, 3 wall, 0.2 layer height

**Actual Results**
only the the last internal wall is under extruded, and the print stop before the real end

**Expected results**
print ok

**Additional Information**
[image]

Predator75, 2018-11-04:
Sorry error in my firmware

closed 2018-11-04 by Predator75

Ghostkeeper, 2019-01-29:
In the UM2 firmware this issue also existed for a while. It happened when the end of the file was being reached in the firmware, which was triggering the printer to stop printing. The printer was done reading the file but there were still a few commands left in the command buffer that would then get cut off.

So maybe a workaround is to add a few comment lines to the bottom of your g-code file so that the firmware can continue reading.

ZhenhuaMao, 2019-01-29:
I found all the ER17 and ER18 errors come from the impact. If you print with 2 type material, the Printcore AA and BB might be printed at the same space. I did the correction for the Ultimaker and it is okay, it looks okay for the first few levels. So, it should be the Cura issue.

Predator75, 2019-02-01:
> 
> 
> Maybe it helps to post the g-code that Cura generates for this cube, then.
I have attached it
[cubo2.zip](https://github.com/Ultimaker/Cura/files/2821387/cubo2.zip)

Ghostkeeper, 2019-02-01:
In your g-code, it specifies that it should draw the last few lines correctly. I see nothing wrong with the g-code.

[image]

What should the g-code specify differently in order to make your printer finish the last few lines of g-code?

ZhenhuaMao, 2019-02-01:
I connected one for the clamps on the connector cable to the AA tubing (should be the BB tubing), might cause the clamp to pull the connector every time the Printcore moving!

ZhenhuaMao, 2019-02-01:
reset the clamp and connector and it is okay now, continuously monitor...