Ultimaker/Cura #4727: [3.5] Weird pattern on the top layer

Liger0, 2018-11-05:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use 0.1 line width and nozzle on printer settings.

**Actual Results**
The top layer is different than the previous ones.

**Expected results**
The top layer should be more circular.

**Additional Information**
[CCR10S_body.zip](https://github.com/Ultimaker/Cura/files/2546572/CCR10S_body.zip)
[image]
[image]

smartavionics, 2018-11-05:
Using Middle slicing tolerance...

[image]

Using Inclusive slicing tolerance...

[image]

Liger0, 2018-11-05:
I didn't think of that setting. Thanks.

closed 2018-11-05 by Liger0